Entware/Entware #498: opkg update reports Error

junyo, 2020-06-30:
opkg update throws errors like this, and I cannot install any package

> Package asterisp, lstaspt-rsk.boxtnel driver. version (null) has no valid architecture, ignoring.
Package asterisk16-chan-ooh323 version 16.10.0-1 has no valid architecture, ignoring.
Package asterisp, lstaspt-playitec nel driver. version (null) has no valid architecture, ignoring.
Package asterisk16-chan-ooh323 version 16.10.0-1 has no valid architecture, ignoring.
Package asterisp, lstaspt-recording.nel driver. version (null) has no valid architecture, ignoring.
Package asterisk16-chan-ooh323 version 16.10.0-1 has no valid architecture, ignoring.
Package asterisp, lstaspt-snoopcnel driver. version (null) has no valid architecture, ignoring.
Package asterisk16-chan-ooh323 version 16.10.0-1 has no valid architecture, ignoring.
Package asterisp, lsta2sd t-h263 version 16.10.0-1 has no valid architecture, ignoring.
Package asterisp, lstunlmonitor asterisk16-age: aste2.ip-app-userevent version 16.12.ipk has no valid architecture, ignoring.
Package asterisp, ld9 version 16:libc, libssp, librt, libpthread, asterisk16 has no valid architecture, ignoring.
Package asterisp, ld9 version 16:libc, libssp, librt, libpthread, asterisk16 has no valid architecture, ignoring.
Package asterisp, lxmppformat-g719 version 16.10.0-1 has no valid architecture, ignoring.
Package asterispecstennel driver. version (null) has no valid architecture, ignoring.
Package asterisk16-chan-ooh323 version 16.10.0-1 has no valid architecture, ignoring.
Package asterisutil-aelparsftnel driver. version (null) has no valid architecture, ignoring.
Package asterisk16-chan-ooh323 version 16.10.0-1 has no valid architecture, ignoring.
Package asterisutil-astdb2sq-16 version 16:libc, libssp, librt, libpthread, asterisk16 has no valid architecture, ignoring.
Package asterisutil-cheec-expr-chanpof stesk16-app-pen-WebRTge: astuserevent version 16:libc, libssp, librt, libpthread, asterisk16 has no valid architecture, ignoring.
Package asterisutil-conf2aeltnel driver. version (null) has no valid architecture, ignoring.
Package asterisk16-chan-ooh323 version 16.10.0-1 has no valid architecture, ignoring.
Package asterisutil-smsqnnel driver. version (null) has no valid architecture, ignoring.
Package asterisk16-chan-ooh323 version 16.10.0-1 has no valid architecture, ignoring.
Package asterisutil-e:reamplayer-chanpof stesk16-app-pen-WebRTge: astuserevent version 16:libc, libssp, librt, libpthread, asterisk16 has no valid architecture, ignoring.
Package asterisAsterrsk.nnel driver. version (null) has no valid architecture, ignoring.
Package rannel drive4.4.188kage: asterisk16-chan-ooh323 version 16.10.0, lo6.1sslaPL-2.0 has no valid architecture, ignoring.
ERROR: truncating field 8 <0x216490> to 21 byteERROR: truncating field 8 <0x21dd68> to 19 byteERROR: truncating field 8 <0x217f90> to 10 byteERROR: truncating field 8 <0x21ce70> to 10 byteERROR: truncating field 8 <0x21ce70> to 6 byteERROR: truncating field 8 <0x21ce70> to 6 byteSegmentation fault

zyxmon, 2020-06-30:
Please give us more info. What is the output of two commands  `cat /opt/etc/entware_release` anf `cat /opt/etc/opkg.conf`

junyo, 2020-06-30:
It is excuted on the ASUS AX3000 router. I've installed aria2 successfully since this happenned.

> \# cat /opt/etc/entware_release
release=entware
arch=arm
cpu=cortex-a9
cpu_subtype=unknown
float=soft
gcc=7.4.0
gcc_flags=-O2 -pipe -march=armv7-a -mtune=cortex-a9 -fno-caller-saves -mfloat-ab                                                                                                                                                             i=soft
libc=glibc
libc_version=2.27

> \# cat /opt/etc/opkg.conf
src/gz entware http://bin.entware.net/armv7sf-k3.2
dest root /
dest ram /opt/tmp
lists_dir ext /opt/var/opkg-lists
option tmp_dir /opt/tmp
arch all 100
arch armv7-3.x 150
arch armv7-3.2 160

zyxmon, 2020-06-30:
I have installed  asterisk16-chan-ooh323 on a fresh Entware installation withou any problems.
My suggestions are
(1) Do a fress reinstall of entware
or
(2) revove all asterisk package with something like
```
opkg list-installed|grep asterisk| sed 's/ - .*$//' | xargs -n 5 opkg remove --force-depends --autoremove
```

junyo, 2020-07-03:
I haven't installed asterisk16-chan-ooh323.  I'll try reinstall entware, Thanks.

closed 2020-07-04 by zyxmon