Entware/Entware #499: Package request: openfortivpn

NeutrinoBR, 2020-06-30:
Package request: openfortivpn
---
openfortivpn is a client for PPP+SSL VPN tunnel services. It spawns a pppd process and operates the communication between the gateway and this process.

- [x] No similar package already added and/or this one got some new killer feature(s),
- [x] Package is in active development (at least, not abandoned for 5 years or more),
- [x] I'm ready to test added package on my mips and get a feedback about it.

---
https://github.com/adrienverge/openfortivpn

ryzhovau, 2020-08-30:
Duplicate of https://github.com/Entware/Entware/issues/327

closed 2020-08-30 by ryzhovau

ryzhovau, 2021-02-13:
We've received 3rd request and gave up. `openfortivpn` added recently.