Entware/Entware #501: Failed to download the package list from bin.entware.net

AlexeyRich, 2020-07-01:
Hello why i am cant download the package from http://bin.entware.net/mipselsf-k3.4/Packages.gz
i am try on 3 different devices and 3 different providers

always answer: 
wget: server returned error: HTTP/1.1 522 Origin Connection Time-out
*** Failed to download the package list from http://bin.entware.net/mipselsf-k3.4/Packages.gz

Collected errors:
 * opkg_download: Failed to download http://bin.entware.net/mipselsf-k3.4/Packages.gz, wget returned 1.

screen https://yadi.sk/d/TCEnObIZhRIuzA

fahaad9, 2020-07-01:
For me it shows 

Collected errors:
 * opkg_download: Failed to download http://bin.entware.net/armv7sf-k2.6/Packages.gz, wget returned 8.

OneCDOnly, 2020-07-01:
Looks like `bin.entware.net` is down again. :disappointed:

Mahtabctg, 2020-07-01:
Can you guys please create a mirror for this site or anyone wants to host one?

pelepelin, 2020-07-01:
> Error 522
Connection timed out

Is there any alternative source for downloading binary packages?

ryzhovau, 2020-07-01:
Repo is back online, sorry for inconvenience.

closed 2020-07-01 by ryzhovau

zloon7, 2024-09-08:
/ # opkg update
Downloading http://bin.entware.net/mipselsf-k3.4/Packages.gz
wget: can't connect to remote host (158.160.45.54): Connection timed out
*** Failed to download the package list from http://bin.entware.net/mipselsf-k3.4/Packages.gz

Downloading http://bin.entware.net/mipselsf-k3.4/keenetic/Packages.gz
wget: can't connect to remote host (158.160.45.54): Connection timed out
*** Failed to download the package list from http://bin.entware.net/mipselsf-k3.4/keenetic/Packages.gz

It is not possible to connect from Ukraine.
I tried from two providers. I tested it via VPN in the browser, there is access. Without VPN - no.
How to solve the problem?