Entware/Entware #502: ifstat 1.1-2 on armv7sf-k3.2 has hard-coded ld-linux.so.3 path(build system leakage)

eigood, 2020-07-07:
The following seems wrong to me.

root@XXX:/jffs/etc/config# opkg install ifstat
Installing ifstat (1.1-2) to root...
Downloading http://bin.entware.net/armv7sf-k3.2/ifstat_1.1-2_armv7-3.2.ipk
Configuring ifstat.
root@XXX:/jffs/etc/config# ldd `which ifstat`
        linux-vdso.so.1 (0x7ef8b000)
        libnetsnmp.so.35 => /opt/lib/libnetsnmp.so.35 (0x76e20000)
        libcrypto.so.1.1 => /opt/lib/libcrypto.so.1.1 (0x76c1b000)
        libnl-3.so.200 => /opt/lib/libnl-3.so.200 (0x76bf6000)
        libm.so.6 => /opt/lib/libm.so.6 (0x76b31000)
        libgcc_s.so.1 => /opt/lib/libgcc_s.so.1 (0x76b14000)
        libc.so.6 => /opt/lib/libc.so.6 (0x769d2000)
        libdl.so.2 => /opt/lib/libdl.so.2 (0x769bf000)
        libpthread.so.0 => /opt/lib/libpthread.so.0 (0x76996000)
        /media/ware3/Entware.2020.05/staging_dir/target-arm_cortex-a9_glibc-2.27_eabi/opt/lib/ld-linux.so.3 => /opt/lib/ld-linux.so.3 (0x76f0c000)
root@XXX:/jffs/etc/config# ifstat
-sh: ifstat: not found

zyxmon, 2020-07-08:
Thanks for reporting. Fixed via https://github.com/Entware/entware-packages/commit/7dd8ac3af5fa7b2c5b30535b5b4d975a4262dec3

closed 2020-07-08 by zyxmon

eigood, 2020-07-08:
That was fast, thanks!  I've gotten use to ifstat on my desktop, as a quick bw monitoring tool for multiple interfaces.  I look forward for the next release.