Entware/Entware #507: Deluge Seg Faults after latest update.

fxPPC, 2020-07-13:
I manage a couple of ASUS routers (a AC68 [kernel 2.6] and a AX56 [kernel 4.1], both with current Merlin-ASUS firmwares), and deluge has stopped working as of the latest update several weeks ago. I’ve hit a brick wall trying to fix it (reinstalling deluge with/without the —autoremove option, reinstalling entware with/without the —force-reinstall option, and even re-initializing the entire router). Everytime I get the same errors about gettext/locale (which are not available to install in entware’s repo), a seg fault, and a error code of 1. Please see the attached screenshot. To re-iterate: the only thing that changed was me typing “opkg update” and then “opkg upgrade”.
[image]

zyxmon, 2020-07-13:
It looks like deluge is broken.

zyxmon, 2020-07-14:
The issue is fixed. Please update your packages.

closed 2020-07-14 by zyxmon

bittles, 2020-07-16:
I'm having a similar problem here.  Installing through entware would break in the same fashion.  Manually touch'ing .configured enabled it to install, but starting it from init.d says the binaries don't exist.

`:/tmp/mnt/sda1/entware/etc/init.d# ./S80deluged start
starting deluged
./S80deluged: line 62: deluged: not found
:/tmp/mnt/sda1/entware/etc/init.d# ./S81deluge-web start
starting deluge-web
./S81deluge-web: line 62: deluge-web: not found
`

`:/tmp/mnt/sda1/entware/bin# ls
ash                        deluge                     fgrep                      htop                       peekfd                     twist3
cftp                       deluge-console             file                       killall                    prtstat                    twistd
cftp3                      deluge-web                 find                       locale.new                 pslog                      twistd3
chardetect                 deluge-web-reset_password  fuser                      localedef.new              pstree                     unrar
chardetect3                deluged                    git                        mailmail                   pyhtmlizer                 xargs
ckeygen                    easy_install               git-receive-pack           mailmail3                  pyhtmlizer3
ckeygen3                   easy_install-3             git-shell                  mako-render                sh
column                     easy_install-3.8           git-upload-archive         nano                       trial
conch                      egrep                      git-upload-pack            netstat                    trial3
conch3                     entware-services           grep                       opkg                       twist
`

zyxmon, 2020-07-17:
@bittles - it looks you have not installed deluge correctly. Python is not installed. Try to
```
opkg update
opkg upgrade
opkg remove --autoremove  deluge-ui-web
opkg install  deluge-ui-web
```
PS ngettext is fixed, although it is a deluge bug that does not case segfault.

bittles, 2020-07-17:
@zyxmon it used to work and broke after an opkg update and upgrade.  After autoremoving deluge and deluge-ui-web, here's my install log with -V 4.
[deluge-ui-web_install.txt](https://github.com/Entware/Entware/files/4939769/deluge-ui-web_install.txt)

Not sure what's going on here at line 425:
`pkg_write_filelist: Creating //opt/lib/opkg/info/deluge-ui-web.list file for pkg deluge-ui-web.
opkg_install_pkg: Resolving conf files for deluge-ui-web
Unknown package '4'.`

zyxmon, 2020-07-17:
It looks like your `/opt/lib/opkg/status` file is broken. It is a text file - try to find package named `4` there and fix it, or reinstall Entware. Another option is to reinstall entware.

bittles, 2020-07-20:
@zyxmon yeah, that was my next step but was trying to avoid it.  Any suggestions on the cleanest way to reinstall entware while making sure the old install isn't left over?  

Amtm didn't seem to have a good way of doing it, and wasn't sure if just formatting the /opt aka /entware sub-directories would clean it properly (symlinks/conf files/etc).  Got my nginx files backed up but figured I'd lose my acme.sh cron renewals that way and have to set all that up again.

bittles, 2020-07-22:
For those finding this later. 

rm /tmp/opt
rm -rf /tmp/mnt/sda1/entware (or just move it to make a backup which is what I did)

Entware reinstalled fine with amtm, deluge installed happily after that,