Entware/Entware #509: Package request: libuuid-devel

WeisMingo, 2020-07-17:
Package request: libuuid-devel
---
Hi. I am currently running DD-WRT Firmware: v3.0-r43516 std. on a TP-Link ARCHER-C7 v5

After installing Entware and Asterisk 16.10.0-1 when a softphone is going to make a call in its CLI an error appears: “No RTP engine was found. Do you have one loaded?” (My list of installed packages and CLI error messages are below)

Here https://forums.asterisk.org/viewtopic.php?t=86518 I found a solution: "yum install uuid uuid-devel libuuid libuuid-devel uuid-c++" with an assumption that adding only libuuid-devel package should be enough to solve the issue.

**Is it possible to add “libuuid-devel” into the Entware set?**

Other sort of help and advices are welcome.
Thanks
______________________________________
- [x] No similar package already added and/or this one got some new killer feature(s),
- [x] Package is in active development (at least, not abandoned for 5 years or more),
- [x] I'm ready to test added package on my TP-Link ARCHER-C7 v5 and make a feedback about it.
______________________________________
My CLI interface error:
Connected to Asterisk 16.10.0 currently running on router (pid = 1938)
[Jul 16 15:51:21] NOTICE[1976]: chan_sip.c:28824 handle_request_subscribe: Received SIP subscribe for peer without mailbox: 506
[Jul 16 15:51:21] NOTICE[1976]: chan_sip.c:25008 handle_response_peerpoke: Peer '506' is now Reachable. (5ms / 600ms)
[Jul 16 15:51:35] ERROR[1976][C-00000001]: rtp_engine.c:489 ast_rtp_instance_new: No RTP engine was found. Do you have one loaded?
[Jul 16 15:51:35] NOTICE[1976][C-00000001]: chan_sip.c:19680 send_check_user_failure_response: RTP init failure for device "506" <sip:506@192.168.0.1>;tag=3132702591 for INVITE, code = -9
router*CLI>

My installed packs:
root@router:~# opkg list-installed
asterisk16 - 16.10.0-1
asterisk16-app-confbridge - 16.10.0-1
asterisk16-bridge-builtin-features - 16.10.0-1
asterisk16-bridge-holding - 16.10.0-1
asterisk16-bridge-native-rtp - 16.10.0-1
asterisk16-bridge-simple - 16.10.0-1
asterisk16-bridge-softmix - 16.10.0-1
asterisk16-chan-iax2 - 16.10.0-1
asterisk16-chan-mgcp - 16.10.0-1
asterisk16-chan-rtp - 16.10.0-1
asterisk16-chan-sip - 16.10.0-1
asterisk16-codec-a-mu - 16.10.0-1
asterisk16-codec-alaw - 16.10.0-1
asterisk16-codec-g722 - 16.10.0-1
asterisk16-codec-g729 - 1.4.3-1
asterisk16-codec-gsm - 16.10.0-1
asterisk16-codec-ilbc - 16.10.0-1
asterisk16-codec-opus - 20171009-1
asterisk16-codec-ulaw - 16.10.0-1
asterisk16-curl - 16.10.0-1
asterisk16-format-g729 - 16.10.0-1
asterisk16-format-gsm - 16.10.0-1
asterisk16-format-h264 - 16.10.0-1
asterisk16-format-ogg-vorbis - 16.10.0-1
asterisk16-format-pcm - 16.10.0-1
asterisk16-format-wav - 16.10.0-1
asterisk16-format-wav-gsm - 16.10.0-1
asterisk16-func-channel - 16.10.0-1
asterisk16-func-config - 16.10.0-1
asterisk16-func-dialgroup - 16.10.0-1
asterisk16-func-dialplan - 16.10.0-1
asterisk16-func-enum - 16.10.0-1
asterisk16-func-env - 16.10.0-1
asterisk16-func-global - 16.10.0-1
asterisk16-func-jitterbuffer - 16.10.0-1
asterisk16-func-uri - 16.10.0-1
asterisk16-func-vmcount - 16.10.0-1
asterisk16-func-volume - 16.10.0-1
asterisk16-pbx-loopback - 16.10.0-1
asterisk16-pbx-realtime - 16.10.0-1
asterisk16-pbx-spool - 16.10.0-1
asterisk16-res-adsi - 16.10.0-1
asterisk16-res-config-mysql - 16.10.0-1
asterisk16-res-http-websocket - 16.10.0-1
asterisk16-res-monitor - 16.10.0-1
asterisk16-res-musiconhold - 16.10.0-1
asterisk16-res-parking - 16.10.0-1
asterisk16-res-pjproject - 16.10.0-1
asterisk16-res-realtime - 16.10.0-1
asterisk16-res-rtp-asterisk - 16.10.0-1
asterisk16-res-rtp-multicast - 16.10.0-1
asterisk16-res-smdi - 16.10.0-1
asterisk16-res-srtp - 16.10.0-1
asterisk16-res-stasis - 16.10.0-1
asterisk16-res-stasis-device-state - 16.10.0-1
asterisk16-res-stun-monitor - 16.10.0-1
asterisk16-res-timing-timerfd - 16.10.0-1
asterisk16-util-check-expr2 - 16.10.0-1
asterisk16-util-muted - 16.10.0-1
asterisk16-util-smsq - 16.10.0-1
asterisk16-util-stereorize - 16.10.0-1
asterisk16-voicemail - 16.10.0-1
bcg729 - 1.0.4-4
ca-bundle - 20190110-2
entware-opt - 227000-3
entware-release - 1.0-2
entware-upgrade - 1.0-1
file - 5.38-2
findutils - 4.7.0-1
glib2 - 2.58.3-5
grep - 3.4-1
jansson - 2.12-1
libattr - 2.4.48-2
libblkid - 2.35.1-1
libbz2 - 1.0.8-1
libc - 2.27-9
libcap - 2.33-2
libcurl - 7.70.0-1
libedit - 20191231-3.1-1
libffi - 3.3-2
libgcc - 8.4.0-9
libiconv-full - 1.11.1-4
libintl-full - 0.19.8.1-2
libltdl - 2.4.6-2
liblzma - 5.2.5-1
libmagic - 5.38-2
libmariadb - 3.1.8-1
libmbedtls - 2.16.6-1
libmount - 2.35.1-1
libncurses - 6.2-1
libncursesw - 6.2-1
libogg - 1.3.4-1
libopenssl - 1.1.1g-1
libopus - 1.3.1-3
libpcre - 8.43-2
libpj - 2.9-4
libpjlib-util - 2.9-4
libpjmedia - 2.9-4
libpjnath - 2.9-4
libpjsip - 2.9-4
libpjsip-simple - 2.9-4
libpjsip-ua - 2.9-4
libpjsua - 2.9-4
libpjsua2 - 2.9-4
libpopt - 1.16-2
libpq - 12.1-4
libpthread - 2.27-9
librt - 2.27-9
libslang2 - 2.3.2-4
libsqlite3 - 3320100-1
libsrtp2 - 2.2.0-1
libssh2 - 1.9.0-2
libssp - 8.4.0-9
libstdcpp - 8.4.0-9
libuuid - 2.35.1-1
libvorbis - 1.3.6-1
libxml2 - 2.9.10-1
locales - 2.27-9
mariadb-common - 1.0-2
mc - 4.8.24-1
nano - 4.9.3-1
opkg - 2020-05-07-f2166a89-1
terminfo - 6.2-1
uuidd - 2.35.1-1
uuidgen - 2.35.1-1
zlib - 1.2.11-3
zoneinfo-asia - 2020a-1
zoneinfo-europe - 2020a-1

ryzhovau, 2020-07-17:
`libuuid-devel` may required on build stage. You may install library header as described [here](https://github.com/Entware/Entware/wiki/Using-GCC-for-native-compilation#headers-installation), but sure problem is somewhere else.

WeisMingo, 2020-07-17:
**ryzhovau**, thanks for your advice. You were right, it didn't help despite after headers' downloading I've fully reinstalled both Entware and Asterisk with its modules. RTP issue is still there.

closed 2020-07-18 by WeisMingo

WeisMingo, 2020-07-19:
... Ultimately I have found the reason. The CLI command 'module load res_rtp_asterisk.so' failed with error 'ast_pjproject_caching_pool_init: symbol not found'. After installing the asterisk16-res-pjproject module RTP service is loading OK.