Entware/Entware #513: Please add docker

skittlesvampir, 2020-07-27:
Package request
---
If we would have a proper docker running on our embedded devices we could do so many fun things. Docker is great and very popular so please add it.

- [x] No similar package already added and/or this one got some new killer feature(s),
- [x] Package is in active development (at least, not abandoned for 5 years or more),
- [x] I'm ready to test added package on my armv7l and get a feedback about it.

osminog, 2020-07-27:
+1 docker package would be great, especially for keenetic routers

skittlesvampir, 2020-07-27:
If you have docker you can do basically whatever you want as long as hardware is good enough.

zyxmon, 2020-07-27:
docker is kernel dependent. Embeded devices have kernels that do not support docker.
NASes that have kernels that support docker have usually docker support from the box. We will not add docker to Entware.

closed 2020-07-27 by zyxmon

skittlesvampir, 2020-07-27:
Lets add the linux kernel to entware😂

zyxmon, 2020-07-27:
Let your device manufacturer compile appropriate kernel. Entware is a userspace binaries repo.

skittlesvampir, 2020-07-27:
I already talked with asustor about that  because I had another program before that had kernel requirements.
They said exactly the following:
> Hello Ben,

> Our NAS is heavily custom Linux. So a lot of functions or package originally that normal Linux has is not installed.

> So, currently, this feature is not supported.

> However, we do respect that it's your system and you have all the freedom to do so.

> But our help will be limited as we have never tested it in our lab, so we can't be sure if it is possible or not nor what kind of impact it will cause to the system stability.

I think they want to make it harder for normies to mess stuff up and make a bad kernel but on the other hand for advanced users, it's harder to do what we want.

leonpano2006, 2023-08-31:
but i think kernel can be modify