Entware/Entware #514: Package request: vsftpd-tls

lelik, 2020-07-29:
Прошу собрать пакет vsftpd-tls для роутеров Keenetic

В текущем репозитории есть пакет vsftpd-ext, но он не поддерживает защищенное соединение TLS
http://pkg.entware-keenetic.ru/binaries/keenle/Packages.html 
Готов протестировать на своем роутере Keenetic Ultra (KN-1810) 
Платформа - MIPSel

The-BB, 2020-07-29:
Для начала, рекомендую обновить систему (Entware-Keenetic => Entware).

lelik, 2020-07-29:
Как бы и нечего обновлять, устанавливался отсюда.
https://forum.keenetic.net/topic/4223-entware-3x-и-entware-ng-объединились-новый-проект-называется-просто-entware/?do=findComment&comment=49963

http://bin.entware.net/mipselsf-k3.4/installer/mipsel-installer.tar.gz
Если только opkg update && opkg upgrade то это тоже делал.

The-BB, 2020-07-29:
Тогда вопрос к Entware team maintainer: какая killer feature y vsftpd-ext (ext - extended?) vs vsftpd (openwrt packages)?

ryzhovau, 2020-07-29:
Насколько могу вспомнить, собирал `vsftpd-ext` из-за нормальной [поддержки](https://vsftpd.devnet.ru/rus/#extensions) кодировок.

closed 2020-08-06 by The-BB