Entware/Entware #523: Package request: RIPE Atlas Software Probe

pasha-zzz, 2020-08-14:
Package request: RIPE Atlas Software Probe
---
"Software probes" are software packages that work as RIPE Atlas probes without the need for any dedicated RIPE Atlas hardware. They are meant to be installed on infrastructure maintained by RIPE Atlas probe hosts themselves - e.g. on virtual machines, home routers, existing servers, etc. Various platforms and operating systems have different levels of support and ease-of-use.

- [x] No similar package already added and/or this one got some new killer feature(s),
- [x] Package is in active development (at least, not abandoned for 5 years or more),
- [x] I'm ready to test added package on my {platform} and get a feedback about it.

https://atlas.ripe.net/docs/software-probe
Github: https://github.com/RIPE-NCC/ripe-atlas-software-probe
Package for Turris routers (as example): https://wiki.turris.cz/doc/en/howto/atlas-probe

ryzhovau, 2020-09-02:
> Package for Turris routers (as example): https://wiki.turris.cz/doc/en/howto/atlas-probe

Looks like a [bunch of shell scripts](https://gitlab.nic.cz/turris/turris-os-packages/-/blob/f7a8cb8caff2e22636d81759a20ac9cad4e019ba/net/atlas-sw-probe/Makefile), which can be used standalone.

Nothing to compile here, sorry.

pasha-zzz, 2020-09-02:
Not compile but add to Entware as IPK with init.d autorun script...

ryzhovau, 2020-09-02:
No worth of packing, sorry.

closed 2020-09-02 by ryzhovau