Entware/Entware #534: Support for Spotlight in Samba

baro77, 2020-09-14:
Gnome Tracker OR Samba 4.12 + Elasticsearch + fscrawler
---
{In MacOS Catalina Time Machine seems to need compulsory spotlight support: current SMB 4.11.6 it can be based only on tracker, Smb 4.12 can rely on elastic+fscrawler  }

- [X ] No similar package already added and/or this one got some new killer feature(s),
- [X] Package is in active development (at least, not abandoned for 5 years or more),
- [X] I'm ready to test added package on my {DD-WRT v3.0-r41954 std (01/09/20) on Netgear R6300v2 w/ external USB3 1TB disk} and get a feedback about it.

---
Please, fill `{}` fields when asking for new package or delete this template in any other case.

ryzhovau, 2020-09-18:
This package came from OpenWrt with no significant changes. Upstream [explicitly disabled](https://github.com/Entware/entware-packages/blob/master/net/samba4/Makefile#L179) spotlight support with some other features to keep SAMBA lightweight and fit to most of embedded devices.

Won't add it, sorry.

closed 2020-09-18 by ryzhovau

VA1DER, 2020-12-05:
OpenWRT and Entware have very different goals.  OpenWRT is intended to produce a complete OS for embedded sytems and those embedded systems, for the most part, can not be relied on to have storage that is even writable.  Everything must fit on the device's internal storage so everything must be as light and small as possible.

Entware is also for embedded systems, but it is for embedded systems with writable mounts and for the vast majority of users this means for all practical purposes there will be no space restrictions.  Following OpenWRT's upstream configuration religiously only has the effect of needlessly removing features from the one package system that people turn to to get MORE features than what their native embedded OS can support.

This argument is especially true for Samba, where most embedded devices already include it, and if people are installing it from Entware it is because they are looking for features their native version doesn't have.

zyxmon, 2020-12-05:
@VA1DER - there are 2500+ packages in Entware now. About 100-200 of packages are Entware ones and the rest come from Openwrt. Enyware developers cannot test thousands of packages after each update. So we keep packages as close to Openwrt ones as possible to make development easy,
Changing Openwrt packages configuration leads to a mess. We have tryed it. Only few packages in Entware are different from upstream. We keep the number of such packages minimal to avoid the mess.