Entware/Entware #535: Can not start amule

lindu0208, 2020-09-16:
**Describe the bug**
[amule]:
 **/opt/etc/init.d/S57amuled start**
 Starting amuled...              done. 
/opt/etc/init.d/rc.func: line 17: 20017 Segmentation fault      (core dumped) $PREARGS $PROC $ARGS > /dev/null 2>&1

**Environment (please complete the following information):**
 - Entware feed: x64-k3.2/  
 - Device: synology ds918+
 - Firmware version: ds6.2.23

I tried to debug /opt/bin/amuled, the result as follow:

**gdb /opt/bin/amuled**
GNU gdb (GDB) 8.3.1
Copyright (C) 2019 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-openwrt-linux".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /opt/bin/amuled...
(No debugging symbols found in /opt/bin/amuled)
(gdb) r
Starting program: /opt/bin/amuled 
warning: File "/opt/lib/libthread_db-1.0.so" auto-loading has been declined by your `auto-load safe-path' set to "$debugdir:$datadir/auto-load".
To enable execution of this file add
	add-auto-load-safe-path /opt/lib/libthread_db-1.0.so
line to your configuration file "/var/services/homes/ulin/.gdbinit".
To completely disable this security protection add
	set auto-load safe-path /
line to your configuration file "/var/services/homes/ulin/.gdbinit".
For more information about this security protection see the
"Auto-loading safe path" section in the GDB manual.  E.g., run from the shell:
	info "(gdb)Auto-loading safe path"
warning: Unable to find libthread_db matching inferior's thread library, thread debugging will not be available.

Program received signal SIGSEGV, Segmentation fault.
0x0000000000000005 in ?? ()
(gdb) bt
#0  0x0000000000000005 in ?? ()
#1  0x00007ffff6daec86 in gconv () from /lib/gconv/UTF-32.so
#2  0x00007ffff72aed5f in ?? () from /opt/lib/libc.so.6
#3  0x00007ffff72ae613 in iconv () from /opt/lib/libc.so.6
#4  0x00007ffff78297a1 in wxMBConv_iconv::GetMBNulLen() const () from /opt/lib/libwx_baseu-2.8.so.0
#5  0x00007ffff7829330 in wxMBConv_iconv::MB2WC(wchar_t*, char const*, unsigned long) const () from /opt/lib/libwx_baseu-2.8.so.0
#6  0x00007ffff7828fad in wxMBConv::ToWChar(wchar_t*, unsigned long, char const*, unsigned long) const () from /opt/lib/libwx_baseu-2.8.so.0
#7  0x00007ffff7828599 in wxMBConv::MB2WC(wchar_t*, char const*, unsigned long) const () from /opt/lib/libwx_baseu-2.8.so.0
#8  0x00007ffff7829a99 in wxMBConv::cMB2WC(char const*) const () from /opt/lib/libwx_baseu-2.8.so.0
#9  0x00007ffff78132ae in ?? () from /opt/lib/libwx_baseu-2.8.so.0
#10 0x00007ffff781386b in wxEntry(int&, char**) () from /opt/lib/libwx_baseu-2.8.so.0
#11 0x000000000043dfc2 in ?? ()
#12 0x00007ffff72adfcb in __libc_start_main () from /opt/lib/libc.so.6
#13 0x000000000044294a in ?? ()
(gdb) 

then I tried  **opkg install  gconv-modules** and still  got the same error:

(gdb) bt
#0  0x0000000000000005 in ?? ()
#1  **0x00007ffff6daec86 in gconv () from /lib/gconv/UTF-32.so**
#2  0x00007ffff72aed5f in ?? () from /opt/lib/libc.so.6
#3  0x00007ffff72ae613 in iconv () from /opt/lib/libc.so.6
#4  0x00007ffff78297a1 in wxMBConv_iconv::GetMBNulLen() const () from /opt/lib/libwx_baseu-2.8.so.0
#5  0x00007ffff7829330 in wxMBConv_iconv::MB2WC(wchar_t*, char const*, unsigned long) const () from /opt/lib/libwx_baseu-2.8.so.0
#6  0x00007ffff7828fad in wxMBConv::ToWChar(wchar_t*, unsigned long, char const*, unsigned long) const () from /opt/lib/libwx_baseu-2.8.so.0
#7  0x00007ffff7828599 in wxMBConv::MB2WC(wchar_t*, char const*, unsigned long) const () from /opt/lib/libwx_baseu-2.8.so.0
#8  0x00007ffff7829a99 in wxMBConv::cMB2WC(char const*) const () from /opt/lib/libwx_baseu-2.8.so.0
#9  0x00007ffff78132ae in ?? () from /opt/lib/libwx_baseu-2.8.so.0
#10 0x00007ffff781386b in wxEntry(int&, char**) () from /opt/lib/libwx_baseu-2.8.so.0
#11 0x000000000043dfc2 in ?? ()
#12 0x00007ffff72adfcb in __libc_start_main () from /opt/lib/libc.so.6
#13 0x000000000044294a in ?? ()
(gdb)

zyxmon, 2020-09-16:
Cannot confirm this bug. Tried amule on x64 Qnap NAS and it runs OK. Please try to update your libs with this - https://github.com/Entware/Entware/wiki#note-about-packages-upgrade or make a fresh install of Entware.

lindu0208, 2020-09-16:
I have tried serveral times and slove this by add **export GCONV_PATH=/opt/lib/gconv** to file **/opt/etc/init.d/S57amuled** according https://github.com/Entware/Entware/blob/9cf57f2c2f1a7e82621b407d9ab9d3bd42332a8a/toolchain/glibc/patches/2.23/0018-eglibc-Cross-building-and-testing-instructions.patch

However, why is 0x00007ffff6daec86 in gconv () from **/lib/gconv/UTF-32.so**  other than **/opt/lib/gconv/UTF-32.so** in my debug result? Could anyone help me? @zyxmon

zyxmon, 2020-09-16:
I have used strace with amuled on x64 system (Qnap NAS). It shows that /lib/gconv/UTF-32.so is accessed. This is a bug. One of Entware developers owns Synology Nas. We will try to fix the bug in case he will confirm segfault.

zyxmon, 2020-09-16:
@lindu0208 - I can build a test package that can probably fix the issue. Do you have time to test the fix?

closed 2020-09-16 by zyxmon

lindu0208, 2020-09-16:
Yes, I can test the fix, please upload it. @zyxmon

zyxmon, 2020-09-17:
@lindu0208 - please temporary modify /opt/etc/opkg.conf and change a line with repo to
`src/gz entware http://bin.entware.net/x64-k3.2/t`
Run
```
opkg update
opkg upgrade
```
Check if amule runs ok.
Rollback opkg.conf modification and report.

lindu0208, 2020-09-17:
Everything is ok now, thanks @zyxmon