Entware/Entware #537: Statically compiled wget for securely installing on systems without TLS

Evidlo, 2020-09-22:
**Describe the solution you'd like**
There are [some embedded systems](https://github.com/Evidlo/remarkable_entware/pull/7) that ship with a wget that doesn't support https.  It would be nice if there was a statically compiled wget available at bin.entware.net that could be downloaded, verified, and then used for the rest of the install script.

The steps in the install script would look like this:

1. Download statically compiled wget (e.g. http://bin.entware.net/armv7sf-k3.2/installer/wget_static)
2. Verify hash of wget_static
3. Install entware over https using wget_static

ryzhovau, 2020-09-23:
Also a bunch of root certificates needed to check initial SSL connection, so it's a bit complicated then a simple static wget.

You may install Entware via HTTP in usual  way and [switch to HTTPS](https://github.com/Entware/Entware/wiki/Using-HTTPS-for-opkg) then.

closed 2020-09-23 by ryzhovau

Evidlo, 2020-09-23:
What about something as simple as:

``` bash
precomputed_hash="c965366080b8d2004cde8e34ebb9f910 -"
hash=$(cat opkg opkg.conf ld-2.27.so libc-2.27.so libgcc_s.so.1 libpthread-2.27.so | md5sum)
if [ "$hash" = "$precomputed_hash"]
then
    echo pass
else
    echo fail
fi
```

Also, is there a reason HTTPS isn't enabled by default in opkg.conf after entware is installed?

ryzhovau, 2020-09-24:
We presume every dependency outside of `/opt` as additional point of failure, so we'll keep installation and basic `opkg` functionality as simple as we can. Also, there's some firmwares where Entware deployed to small NAND partition and we have keep basic number of files in light weight.

This behavior was a result of 3 Entware generations and we see no sense to change it in future, sorry.

ryzhovau, 2020-09-24:
> Download statically compiled wget (e.g. http://bin.entware.net/armv7sf-k3.2/installer/wget_static)

BTW, we promise to keep every file of `http://bin.entware.net/armv7sf-k3.2/installer/*` in current state till the end on feed life time, including gcc/libc version numbers, because of some popular firmwares with included their own installation scripts (such as asuswrt-merlin, tomato, padavan and so on), so you may rely on it too and make your own installation script for reMarkable Tablet.