Entware/Entware #539: php7-mod-gd

GitzJoey, 2020-09-25:
Hi,

i'm trying to install php7-mod-gd in RT-AC88U version 384.19
but having error

> admingj@RT-AC88U-D8E0:/tmp/home/root# php-cli -v
> PHP Warning:  PHP Startup: Unable to load dynamic library 'gd.so' (tried: /opt/lib/php/gd.so (libjpeg.so.62: cannot open shared object file: No such file or directory), /opt/lib/php/gd.so.so (/opt/lib/php/gd.so.so: cannot open shared object file: No such file or directory)) in Unknown on line 0
> PHP 7.4.9 (cli) ( NTS )
> Copyright (c) The PHP Group
> Zend Engine v3.4.0, Copyright (c) Zend Technologies
> admingj@RT-AC88U-D8E0:/tmp/home/root# opkg install libjpeg
> Package libjpeg (9c-2) installed in root is up to date.
> admingj@RT-AC88U-D8E0:/tmp/home/root# opkg install libjpeg62
> Unknown package 'libjpeg62'.
> Collected errors:
>  * opkg_install_cmd: Cannot install package libjpeg62.
> admingj@RT-AC88U-D8E0:/tmp/home/root#

is there anyway to install libjpeg.so.62 ?

zyxmon, 2020-09-25:
Not confirmed
```
$ opkg install php7-mod-gd
Installing php7-mod-gd (7.4.9-1) to root...
Downloading http://bin.entware.net/armv7sf-k3.2/php7-mod-gd_7.4.9-1_armv7-3.2.ipk
Installing libpcre2 (10.35-2) to root...
Downloading http://bin.entware.net/armv7sf-k3.2/libpcre2_10.35-2_armv7-3.2.ipk
Installing php7 (7.4.9-1) to root...
Downloading http://bin.entware.net/armv7sf-k3.2/php7_7.4.9-1_armv7-3.2.ipk
Installing libjpeg-turbo (2.0.5-1) to root...
Downloading http://bin.entware.net/armv7sf-k3.2/libjpeg-turbo_2.0.5-1_armv7-3.2.ipk
Installing libpng (1.6.37-7) to root...
Downloading http://bin.entware.net/armv7sf-k3.2/libpng_1.6.37-7_armv7-3.2.ipk
Installing libwebp (1.1.0-2) to root...
Downloading http://bin.entware.net/armv7sf-k3.2/libwebp_1.1.0-2_armv7-3.2.ipk
Installing libfreetype (2.10.2-1) to root...
Downloading http://bin.entware.net/armv7sf-k3.2/libfreetype_2.10.2-1_armv7-3.2.ipk
Installing libtiff (4.1.0-3) to root...
Downloading http://bin.entware.net/armv7sf-k3.2/libtiff_4.1.0-3_armv7-3.2.ipk
Installing libgd-full (2.3.0-3) to root...
Downloading http://bin.entware.net/armv7sf-k3.2/libgd-full_2.3.0-3_armv7-3.2.ipk
Configuring libpng.
Configuring libpcre2.
Configuring libwebp.
Configuring libjpeg-turbo.
Configuring libtiff.
Configuring libfreetype.
Configuring libgd-full.
Configuring php7.
Configuring php7-mod-gd.
$ opkg install php7-cli
Installing php7-cli (7.4.9-1) to root...
Downloading http://bin.entware.net/armv7sf-k3.2/php7-cli_7.4.9-1_armv7-3.2.ipk
Configuring php7-cli.
$ php-cli -v
PHP 7.4.9 (cli) ( NTS )
Copyright (c) The PHP Group
Zend Engine v3.4.0, Copyright (c) Zend Technologies

```

closed 2020-09-25 by zyxmon

GitzJoey, 2020-09-25:
but...but

> 
> admingj@RT-AC88U-D8E0:/tmp/home/root# opkg remove php7-mod-gd
> Removing package php7-mod-gd from root...
> admingj@RT-AC88U-D8E0:/tmp/home/root# opkg install php7-mod-gd
> Installing php7-mod-gd (7.4.9-1) to root...
> Downloading http://bin.entware.net/armv7sf-k2.6/php7-mod-gd_7.4.9-1_armv7-2.6.ipk
> Configuring php7-mod-gd.
> admingj@RT-AC88U-D8E0:/tmp/home/root#

anyway to check ?

zyxmon, 2020-09-25:
probably some libs were not updated - https://github.com/Entware/Entware/wiki#note-about-packages-upgrade

GitzJoey, 2020-09-25:
ok resolved.

admingj@RT-AC88U-D8E0:/tmp/home/root# opkg install libjpeg-turbo
Installing libjpeg-turbo (2.0.5-1) to root...
Downloading http://bin.entware.net/armv7sf-k2.6/libjpeg-turbo_2.0.5-1_armv7-2.6.ipk
Configuring libjpeg-turbo.
admingj@RT-AC88U-D8E0:/tmp/home/root# php-cli -v
PHP 7.4.9 (cli) ( NTS )
Copyright (c) The PHP Group
Zend Engine v3.4.0, Copyright (c) Zend Technologies
admingj@RT-AC88U-D8E0:/tmp/home/root#