Entware/Entware #540: syntax error: unexpected ")" OR not found <any_package>

diman82, 2020-09-26:
**Describe the bug**
nano OR tmux: I get:
`syntax error: unexpected ")"` when running _tmux_ and
`/opt/bin/nano: /opt/bin/nano: line 1: line 1: ▒0a,▒m4Yn1: not founELF▒▒R@@8▒@8: not found` when running _nano_

**Environment (please complete the following information):**
 - Entware feed: don't know
 - Device: RT-AX56U
 - Firmware version: asuswrt-merlin, 384.18

zyxmon, 2020-09-26:
Cannot confirm. Please try to reinstall nano.

closed 2020-09-26 by zyxmon

diman82, 2020-09-26:
> Cannot confirm. Please try to reinstall nano.

Like I said - nano was only an example.. The same exceptions are with EVERY package I try to run..

diman82, 2020-09-26:
Ok, reinstalled entware completely - now all works fine..