Entware/Entware #543: Chrony (for MIPS)

sumerland, 2020-09-28:
Chrony has recently been made available for aarch64 (e.g. Asus RT-AC86U) - many thanks for that!

However, I cannot install the same package via opkg on mips (Asus RT-N66U)

Is that a lack of competence on my end or is the package actually missing?

ryzhovau, 2020-09-29:
Chrony available for every modern Entware feeds.

My guess, you've got Entware-ng, which is abandoned for now.

sumerland, 2020-09-29:
Your are right - opkg.conf points towards
`src/gz entware-ng http://pkg.entware.net/binaries/mipsel`

whereas on my aarch64 router the looks at
`src/gz entware http://bin.entware.net/aarch64-k3.10`

The configuration file is automatically created by the amtm script, which comes along with John's fork of Merlin's Asus firmware. I will contact the developers.

closed 2020-09-29 by sumerland