Entware/Entware #545: libcrypto potentially compiled with hf

kiwiz, 2020-09-30:
**Describe the bug**
[package name]: When trying to invoke `pip3`, the process crashes with a `SIGILL`. Running via GDB shows the following offending instruction:
```
Program received signal SIGILL, Illegal instruction.
0x40a9c568 in ?? () from /opt/lib/libcrypto.so.1.1
(gdb) x/5i $pc-12
   0x40a9c55c:	nop	{0}
   0x40a9c560:	vorr	q0, q0, q0
   0x40a9c564:	bx	lr
=> 0x40a9c568:	mrrc	15, 1, r0, r1, cr14
   0x40a9c56c:	bx	lr
```
`mrrc` seems to be a floating point related instruction, so my guess is this library might've been compiled with hardfloat support? Lemme know if there's any other info I can provide. 

**Environment (please complete the following information):**
 - Entware feed: `armv7sf-k2.6`
 - Device: Asus RT-AC68U
 - Firmware version asuswrt-merlin 384.19

zyxmon, 2020-09-30:
Tried to run pip on a similar hardware - no problems:
```
 pip install --upgrade pip setuptools
Collecting pip
  Downloading pip-20.2.3-py2.py3-none-any.whl (1.5 MB)
     |████████████████████████████████| 1.5 MB 50 kB/s
Collecting setuptools
  Downloading setuptools-50.3.0-py3-none-any.whl (785 kB)
     |████████████████████████████████| 785 kB 2.6 MB/s
Installing collected packages: pip, setuptools
  Attempting uninstall: pip
    Found existing installation: pip 20.1.1
    Uninstalling pip-20.1.1:
      Successfully uninstalled pip-20.1.1
  Attempting uninstall: setuptools
    Found existing installation: setuptools 47.1.0
    Uninstalling setuptools-47.1.0:
      Successfully uninstalled setuptools-47.1.0
Successfully installed pip-20.2.3 setuptools-50.3.0

```

kiwiz, 2020-10-01:
Thanks for investigating. I ran a `fsck` on the drive and came up with a bunch of errors. Reformatting the device fixed it, so it looks like it was just a FS corruption issue. Apologies for the false alarm.

closed 2020-10-01 by kiwiz