Entware/Entware #546: Package Update: nginx-extras

camkidd, 2020-10-05:
Package Update request for "nginx-extras" in rtndev.
Not sure if I should submit pull request to rtndev or just request here?

Anyway, the regular nginx & nginx-ssl packages are both on 1.19.2, but nginx-extras still on 1.18.0

Thanks.

ryzhovau, 2020-10-05:
`nginx-extras` is came from Debian. It preserves the same configs, patches and addons as upstream do.
So, we'll stays the same version [as in Debian](http://ftp.debian.org/debian/pool/main/n/nginx/).

closed 2020-10-05 by ryzhovau

camkidd, 2020-10-05:
Fair enough. Sorry to bother you. 
The nginx makefile references "all-modules" - Is there a package from that, or is that only if I compile myself with make menuconfig?

ryzhovau, 2020-10-05:
No, it's about [nginx from OpenWrt](https://github.com/openwrt/packages/blob/master/net/nginx/Config_ssl.in), which isn't related to `nginx-extras`.

camkidd, 2020-10-05:
sorry I wasn't clear, I did mean that nginx package from ooenwrt as you state - how do I get nginx modules with that package (or nginx-ssl) as opposed to using the Debian version.

ryzhovau, 2020-10-06:
`nginx-ssl` from OpenWrt built with [all available modules](https://github.com/Entware/Entware/blob/master/configs/mipsel-3.4.config#L6320) (excluding UBUS). They are part of nginx binary, not separate loadable files. Check ` nginx -V` for more details.

camkidd, 2020-10-06:
Ahhh! Penny drops!. Thankyou very much for replying.