Entware/Entware #547: run entware-setup or opkg upgrade errors

tsingui, 2020-10-06:
Hi Developers,

**Describe the bug**
[package name]: entware-opt_227000-3_all.ipk

**run entware-setup or 1)opkg update 2)opkg upgrade:**

Upgrading entware-opt on ram from 222108-6 to 227000-3...
Downloading http://bin.entware.net/armv7sf-k2.6/entware-opt_227000-3_all.ipk
Collected errors:
 * opkg_install_pkg: **Package entware-opt md5sum mismatch**. Either the opkg or the package index are corrupt. Try 'opkg update'.

**Environment :**
 - Entware feed: [armv7sf-k2.6]
 - Device [Asus RT-AC56U router]
 - Firmware version [asuswrt-merlin384.6]

So thank you for all your hard work!

ryzhovau, 2020-10-06:
Looks like you trying to upgrade Entware-ng to Entware and something went wrong. Please, install Entware from scratch.

closed 2020-10-06 by ryzhovau

zyxmon, 2020-10-06:
May be similar to this one - https://github.com/Entware/Entware/issues/74