Entware/Entware #549: python3-pynacl

ivankravets, 2020-10-17:
Please add python3-pynacl to opkg registry. Home Assistant depends on it. Thanks!

zyxmon, 2020-10-18:
Please, try to install it yourself - https://github.com/Entware/Entware/wiki/Self-installation-of-python-modules
PS There are no dev packages in openwrt and entware. You need to install headers manually and a nondev package.

ivankravets, 2020-10-18:
> You need to install headers manually

What is the best practice on how to do it? I don't want to break OS. Thanks!

zyxmon, 2020-10-18:
Just follow the wiki.

ivankravets, 2020-10-18:
Thanks!

if someone will need a quick guide:

1. Download `include.tar.gz` as described at https://github.com/Entware/Entware/wiki/Using-GCC-for-native-compilation#headers-installation
2. Unpack on local machine, copy `ffi.h` to `/opt/include/ffi.h`
3. `ln -s /opt/lib/libffi.so.7 /opt/lib/libffi.so` on a target device

Now, use the classic `pip` tool.

closed 2020-10-18 by ivankravets