Entware/Entware #552: Please update chrony from 3.5 to 4.0

JGrana01, 2020-10-25:
Chrony has been update to 4.0. It adds a number of improvements and a new Network Time Security option.
I have compiled 4.0 but cannot get it to add NTS - it requires gnutls. The configuration uses pkg-config to check for gnutls - I have installed both gnutls modules using opkg on an ASUS AX88U and also installed pkg-config. It still does not add/compile gnutls into chronyd. There are no ".pc" files in it's search path. As a result, it doesn't believe gnutls is installed.

This security feature would be useful on embedded devices. 

- [X ] No similar package already added and/or this one got some new killer feature(s),
- [X ] Package is in active development (at least, not abandoned for 5 years or more),
- [X ] I'm ready to test added package on my platform [e.g. mipselsf-k3.4] and get a feedback about it.

butaford, 2020-10-26:
I'm not a developer, but I put together a package for your. Try it.

[chrony_ipk_entware.zip](https://github.com/Entware/Entware/files/5437895/chrony_ipk_entware.zip)
[chrony_src_entware.tar.gz](https://github.com/Entware/Entware/files/5437896/chrony_src_entware.tar.gz)

JGrana01, 2020-10-26:
Nice, thanks. I even see nettle!
Unfortunately, I am on an Asus RT-AX88U which uses aarch64-k3.10
Any chance you have that build environment?
Happy to test if you do!

butaford, 2020-10-26:
aarch64:
[chrony_ipk_entware.zip](https://github.com/Entware/Entware/files/5439331/chrony_ipk_entware.zip)

JGrana01, 2020-10-26:
Thanks, installed fine. The one issue is that NTS is still disabled. The chrony configure script doesn't think we have gnutls so it disables NTS:

chronyd -v

chronyd (chrony) version 4.0 (+CMDMON +NTP +REFCLOCK -RTC +PRIVDROP -SCFILTER -SIGND +ASYNCDNS **-NTS** -SECHASH +IPV6 -DEBUG)

I looked at the configure script. It uses pkg-config to determine if gnutls is present. Since we have no gnutls.pc file, it fails (silently) and does not add support for gnutls (which NTS requires).

Thanks again.

ryzhovau, 2020-10-26:
This package was taken from upstream with no any changes https://github.com/openwrt/packages/tree/master/net/chrony

So chrony 4.0 will ariive here during next sync, which is planned in next couple of weeks.

closed 2020-10-26 by ryzhovau

JGrana01, 2020-10-26:
Thanks ryzh0vau. BTW butaford, I installed the chrony-gnutls_4.0-1_aarch64-3.10.ipk after uninstalling chrony. It does now appear that NTS is working:
chronyd (chrony) version 4.0 (+CMDMON +NTP +REFCLOCK -RTC +PRIVDROP -SCFILTER -SIGND +ASYNCDNS +NTS -SECHASH +IPV6 -DEBUG)

I will do some testing with NTS

butaford, 2020-10-26:
i forgot to add ca-bundle package as dependency

jackyaz, 2020-12-05:
It looks like the version for aarch64 (RT-AC86U) is missing NTS support:
[image]