Entware/Entware #553: Could you please add ipmitool package

djbobyd, 2020-10-28:
Hello,
I would like to first thank you for this nice project and all the effort you put in it. 
I would like to ask if it is possible to include an "ipmitool" package?
It is very useful if you want to configure your router to start remotely your servers. For example, I have the following usecase:
I have 2 servers connected to UPS. However it often happens that the electricity is out for a longer period and teh ups cannot maintain the power for this long period. As a result the servers receive a shutdown command. Unfortunately in this case when the power is back again they cannot start automatically. Both servers have IPMI interface and I can start them with ipmitool. I would like to automate this step so that my servers start after a power failure even when I am not at home.

- [X] No similar package already added and/or this one got some new killer feature(s),
- [X] Package is in active development (at least, not abandoned for 5 years or more),
- [X] I'm ready to test added package on my platform [e.g. mipselsf-k3.4] and get a feedback about it.

Thanks a lot in advance

butaford, 2020-10-31:
The package is already present in the Entware source codes https://github.com/Entware/entware-packages/tree/master/admin/ipmitool, but for some reason the developers did not include it in the assembly https://github.com/Entware/Entware/blob/master/configs/mipsel-3.4.config#L2418.
It remains only to collect it. :-)

Try
[ipmitool_1.8.18-5_mipsel-3.4.ipk.zip](https://github.com/Entware/Entware/files/5468632/ipmitool_1.8.18-5_mipsel-3.4.ipk.zip)
[ipmitool_1.8.18-5_mips-3.4.ipk.zip](https://github.com/Entware/Entware/files/5468640/ipmitool_1.8.18-5_mips-3.4.ipk.zip)
[ipmitool_1.8.18-5_aarch64-3.10.ipk.zip](https://github.com/Entware/Entware/files/5468641/ipmitool_1.8.18-5_aarch64-3.10.ipk.zip)

djbobyd, 2020-11-10:
Thanks for the links, but could you please provide me a link for armv7. 
I tried to figure out the link but couldn't.
Thanks!

butaford, 2020-11-10:
Comment updated: https://github.com/Entware/Entware/issues/553#issuecomment-719896756

djbobyd, 2020-11-10:
It works as expected. Thanks a lot for the links. 
All I need now is to have it available through the official list so that I can get any updates when they come. 
How can I find out why it wasn't included in the assembly? Is there a place where I can get more info or should I wait for someone from the devs to respond?

Thanks!

ryzhovau, 2020-11-10:
Thanks for feedback. I'll add `ipmitool` to repo soon.

closed 2021-04-04 by ryzhovau