Entware/Entware #556: lz4 binary

benneti, 2020-11-04:
It would be nice to have a package for the binary of lz4 ([https://github.com/lz4/lz4/](https://github.com/lz4/lz4/)).

There is already a package available for the library [entware-packages/blob/master/libs/liblz4/Makefile](https://github.com/Entware/entware-packages/blob/master/libs/liblz4/Makefile) and it looks like it defines instructions for lz4 as well as liblz4 but on https://bin.entware.net/armv7sf-k3.2/ I can only find the liblz4.
This is also true on the device with
``` 
$ opkg -v
opkg version f2166a89b7e96e2c60002959731eebc5f45fa318 (2020-05-07)
$ uname -a
Linux reMarkable 4.14.78 #1 SMP PREEMPT Tue Sep 22 21:30:00 UTC 2020 armv7l GNU/Linux
```
.
It would be nice if you could include the package in line [armv7-3.2.config#L7002](https://github.com/Entware/Entware/blob/4fa08a36855ef9185eae017ba8977a28dba9e615/configs/armv7-3.2.config#L7002).

- [ ] No similar package already added and/or this one got some new killer feature(s),
  (There are other compression algorithms but lz4 is suited for very fast compression)
- [x] Package is in active development (at least, not abandoned for 5 years or more),
- [x] I'm ready to test added package on my platform [e.g. mipselsf-k3.4] and get a feedback about it.

butaford, 2020-11-06:
[lz4_1.9.2-4_mips-3.4.ipk.zip](https://github.com/Entware/Entware/files/5500916/lz4_1.9.2-4_mips-3.4.ipk.zip)
[lz4_1.9.2-4_mipsel-3.4.ipk.zip](https://github.com/Entware/Entware/files/5500917/lz4_1.9.2-4_mipsel-3.4.ipk.zip)
[lz4_1.9.2-4_x64-3.2.ipk.zip](https://github.com/Entware/Entware/files/5500918/lz4_1.9.2-4_x64-3.2.ipk.zip)
[lz4_1.9.2-4_x86-2.6.ipk.zip](https://github.com/Entware/Entware/files/5500919/lz4_1.9.2-4_x86-2.6.ipk.zip)
[lz4_1.9.2-4_aarch64-3.10.ipk.zip](https://github.com/Entware/Entware/files/5500920/lz4_1.9.2-4_aarch64-3.10.ipk.zip)

benneti, 2020-11-09:
thank you for the links.
1) They work (at least for armv7) no problems so far!
2) How can I find these links myself to stay up to date (as long as they are not included in the accessible repo)?

ryzhovau, 2020-11-10:
Binaries will be accessible after 1-2 weeks, after next sync with OpenWrt.

benneti, 2020-11-10:
Awesome, thank you!

token0, 2021-02-26:
> [lz4_1.9.2-4_mips-3.4.ipk.zip](https://github.com/Entware/Entware/files/5500916/lz4_1.9.2-4_mips-3.4.ipk.zip)
> [lz4_1.9.2-4_mipsel-3.4.ipk.zip](https://github.com/Entware/Entware/files/5500917/lz4_1.9.2-4_mipsel-3.4.ipk.zip)
> [lz4_1.9.2-4_x64-3.2.ipk.zip](https://github.com/Entware/Entware/files/5500918/lz4_1.9.2-4_x64-3.2.ipk.zip)
> [lz4_1.9.2-4_x86-2.6.ipk.zip](https://github.com/Entware/Entware/files/5500919/lz4_1.9.2-4_x86-2.6.ipk.zip)
> [lz4_1.9.2-4_aarch64-3.10.ipk.zip](https://github.com/Entware/Entware/files/5500920/lz4_1.9.2-4_aarch64-3.10.ipk.zip)
> [lz4_1.9.2-4_armv7-3.2.ipk.zip](https://github.com/Entware/Entware/files/5501584/lz4_1.9.2-4_armv7-3.2.ipk.zip)

Thanks, @benneti !

closed 2021-04-04 by ryzhovau