Entware/Entware #558: syncthing package startup script makes its UI listen on ALL interfaces

VA1DER, 2020-11-12:
syncthing package startup script makes its UI listen on ALL interfaces

**DESCRIPTION**
syncthing and syncthing_nohf: The startup script for the syncthing package, S92syncthing, uses sed to replace the listen address for syncthing's UI in the configuration from 127.0.0.1 to 0.0.0.0.  This causes syncthing to listen on ALL interfaces.  Since entware is often used on routers, this is a terrible huge breach of security as it will expose syncthing's UI to the entire internet and syncthing does not have a password set by default.

Even worse is that the startup script does this every time syncthing is run, not just the first time.  So a person going in and manually changing this setting back may not know that the next time the system starts up it will do this again.

There is a reason why Syncthing defaults to loopback.  Leave it as an exercise for the person installing to set up the GUI listen address.

zyxmon, 2020-11-12:
This was suggested by one of Entware users. In case you have a fix, that works for many platforms, please make a PR.

closed 2020-11-12 by zyxmon

VA1DER, 2020-11-14:
> This was suggested by one of Entware users.
This doesn't make it a good idea.  Perhaps you don't know, but it is not the Syncthing protocol you are making Syncthing listen to on all interfaces, but the Syncthing UI!  And there is no default password.

As I mentioned, this is a huge security problem.  For all users of dd-wrt, tomato, and any device which potentially has access to the internet, this leaves a gaping hole in your security by letting the entire internet have access to Syncthing's control UI, and it does it silently and it does it every time Syncthing is run.

It already is not a good idea to do this automatically at first run.  It is irresponsible and a violation of the user's choice to do this at every run.  What if the user runs Syncthing, looks at the listen address and sees the security hole and changes this back to 127.0.0.1?  The change will stick while Syncthing is running, but then get silently changed back the next time the device reboots.  Even users who have seen and changed it back will likely not know this script is going to do it again.  This is spectacularly dumb, and I'm sorry there is no nicer way to put this.

> In case you have a fix, that works for many platforms
The fix is easy for every platform, and that is to remove the script that does this.

The proper way to do something like this is not at run-time every run, but at installation time with:

- Education.  A simple message telling the user how to create the syncthing config file and how to make the Syncthing UI listen on external ports if they want, or how to use ssh to create a safe tunnel.
- Education plus a helper script.  Telling the user the above and also installing a script to do this editing for the user and telling them "Use syncthing-listen to change the listen port for Syncthing's UI".  In the usage for syncthing-listen have it tell the user the options and make the user actually type in the listen address they want "syncthing-listen 192.168.1.1", or "syncthing-listen 0.0.0.0" and then if it's 0.0.0.0 issue a warning message telling the user the ramifications and asking them if they are sure.

The responsible action is also to issue a vulnerability report so that people who unknowingly have this security hole have a chance at fixing it.  For dd-wrt and tomato for sure (I don't know for other appliances), entware is almost ubiquitous.  And given how popular syncthing is becoming, the user base is large enough to make this the responsible thing to do.  I will be making reports to CCIRC, CERT/CC, and US-CERT.  It would be nice to include in the report that this is fixed by entware.

zyxmon, 2020-11-14:
Please make a PR after testing your solution.