Entware/Entware #559: GCC fails on Tomato firmware with "ilegal instruction"

VA1DER, 2020-11-17:
**Description**
[gcc]: gcc fails to run, issuing "ilegal instruction" error after it was installed according to wiki instructions.

```
root@tusker:/opt/root# gcc -o helloworld helloworld.c 
gcc: internal compiler error: Segmentation fault (program cc1)
Please submit a full bug report,
with preprocessed source if appropriate.
See <https://dev.openwrt.org/> for instructions.
Illegal instruction

```

**Environment:**
 - Entware feed: arm7armv7sf-k2.6
 - Device: Netgear R7000
 - Firmware version: FreshTomato 2020.6 AIO

ryzhovau, 2020-11-17:
Looks like some incompatibility between CPU and toolchain. Can't help there.

closed 2020-11-17 by ryzhovau

zyxmon, 2020-11-17:
IMHO it is due to some missing syscall in Tomato kernel. There are some problems with hardware using kernel 2.6 that cannot be solved.

VA1DER, 2020-11-19:
I would tend to agree with zyxmon.  I just was able to get it running on another R7000 with DD-WRT kernel 4.4.242.  I suspect there is a way to get this to work on kernel 2.6 but cross compiling for targets with vastly different kernels isn't something I have expertise in and I'm not sure it's worth it for a ten year old kernel.  I am likely going to migrate all my personal FreshTomato installs to DD-WRT, so I don't really have the motivation to chase this down myself.

I would recommend, though, that the Wiki page be modified to indicate that kernel 2.6 isn't supported for GCC and that the package be removed from the Entware on kernel 2.6 repository.

Someone needs to talk to the FT decision makers and twist their arm into working with DD-WRT to get a modern kernel.

zyxmon, 2020-11-19:
Kernel 2.6 is supported for GCC. I have just compiled HellWorld example on an old tomato arm router:
```
Tomato Firmware 1.28.0000 --130_2 K26ARM USB AIO-64K

Linux kernel 2.6.36.4brcmarm and Broadcom Wireless Driver 6.37.14.86 (r456083)
Copyright (C) 2013-2014 Tomato-ARM Team
```
Some binaries may fail (gcc is just an example) on some firmwares with 2.6. This is due to patches in 2.6 kernel and/or kernel configuration. Entware uses generic 2.6 kernel and configuration in it toolchain to build binaries. There are usually no problems with asuswrt firmwares, there may be problems with tomato and dd-wrt.

VA1DER, 2020-11-19:
Tomato's kernel hasn't changed in a decade.  I am using the exact same kernel today, and with all likelihood it isn't just the same version but the same binary.  What version of GCC did you just test... was it the latest from entware?

zyxmon, 2020-11-20:
I have tested the current version from the repo.
Plz note - not only kernel sources matter but also the kernel config used in firmware. Firmware developers may have disabled some options to save space in your case. That will create a missing syscall that is used in Entware toolchain.

VA1DER, 2020-12-05:
The latest (Dec 3rd 2020) update to Entware has fixed the issue.  There appears to have been no change to gcc, so the next likely candidate was the update to libc.

zyxmon, 2020-12-05:
libc was not updated, binutils in the toolchain was updated. Dependant package binutils was also updated.

VA1DER, 2020-12-06:
Interesting because on today's upgrade:
```
root@guardian:/tmp/mnt/sdc1# opkg upgrade
Upgrading libc on root from 2.23-10 to 2.23-11...
Downloading http://bin.entware.net/armv7sf-k2.6/libc_2.23-11_armv7-2.6.ipk
...
Configuring libc.
...
```
I see that libgcc was also on the list of upgraded packages.  I figured this was just because libc was, but I don't know.  In any case, gcc is working now.  If I'd have been smarter, I'd have tried it through gdb when it failed to determine exactly where the bad instruction was.  Moot now.