Entware/Entware #560: 'tang' package missing

kinafu, 2020-11-21:
I could not find the package [`tang`, which is already present in the git repository](https://github.com/Entware/entware-packages/tree/master/utils/tang) in the [bin.entware.net package list](http://bin.entware.net/armv7sf-k2.6/Packages.html).  
(And so the package manager is not able to install `tang` as well.)

 - Entware feed: `armv7sf-k2.6`

butaford, 2020-11-21:
https://github.com/Entware/Entware/blob/df9ebe38927c7c21a5de9082c1db9d0896fce157/configs/armv7-2.6.config#L2510

butaford, 2020-11-22:
try this (dependencies within the archive)
[tang_7-1_armv7-2.6.zip](https://github.com/Entware/Entware/files/5579678/tang_7-1_armv7-2.6.zip)

kinafu, 2020-12-28:
Thanks for providing a binary.

I tried to install the package by doing

``` sh
mkdir tang && cd tang
opkg update
opkg upgrade
wget https://github.com/Entware/Entware/files/5579678/tang_7-1_armv7-2.6.zip
unzip tang_7-1_armv7-2.6.zip
opkg print-architecture
opkg install *.ipk
```

The installation process returned:
```
Configuring libncursesw.
Configuring libncurses.
Configuring libreadline.
Configuring bash.
Configuring jansson.
Configuring zlib.
Configuring libopenssl.
Configuring libjose.
Configuring libhttp-parser.
Configuring xinetd.
Configuring jose.
Configuring tang.
//opt/lib/opkg/info/tang.postinst: line 13: /opt/libexec/tangd-update: not found
Collected errors:
 * pkg_run_script: package "tang" postinst script returned status 127.
 * opkg_configure: tang.postinst returned 127.
```

The installation seems to be incomplete. Maybe a dep is missing?

kinafu, 2021-03-11:
I noticed that there has been a change in the [tang code in the entware-packages ](https://github.com/Entware/entware-packages/commit/d0cb262ea1e6579e44f27dbc83f82f6a62d896f3#diff-5535c310ca07b7c893f39aea7055e5b4dccb21c1eeff27e324ecc1620306578c)repo.

Is there anyone with an idea about the missing package uponm the `tang` installation? @butaford maybe?

closed 2021-04-04 by ryzhovau

kinafu, 2021-06-26:
The installation of the binary, which you sent, succeeded on my router.

But the tang server did not start after installation. So I looked up how to manually start the `tang` server:
``` sh
echo  "" > /jffs/syslog.log
/tmp/mnt/sda1/entware/etc/init.d/S10xinetd start
# Starting xinetd...              done.
tail -f /jffs/syslog.log
# Jun 26 14:14:55 admin: Started xinetd from .
# Jun 26 12:14:56 xinetd[1042]: Unknown user: root [file=/opt/etc/xinetd.d/tangdx] [line=6]
# Jun 26 12:14:56 xinetd[1042]: Error parsing attribute user - DISABLING SERVICE [file=/opt/etc/xinetd.d/tangdx] [line=6]
# Jun 26 12:14:56 xinetd[1042]: service/protocol combination not in /etc/services: tangd/tcp
# Jun 26 12:14:56 xinetd[1042]: 1042 {init_services} no services. Exiting...
```

I tried to debug the errors.

1. `Jun 26 12:14:56 xinetd[1042]: Unknown user: root [file=/opt/etc/xinetd.d/tangdx] [line=6]`
- I changed the line `user = root` to `user = admin`
- ✅ Result: The error does not appear anymore in the log after executing the xinetd start command once more
2. `Jun 26 12:14:56 xinetd[1042]: service/protocol combination not in /etc/services: tangd/tcp`
- ❓ I could not figure out how to add the tang service to the `services` file. In my firmware that file is located on the `/rom` partition and in read-only mode.

zyxmon, 2021-06-26:
>> I could not figure out how to add the tang service to the services file. In my firmware that file is located on the /rom partition and in read-only mode.

You can use xinetd like this - https://azouhr.wordpress.com/2012/06/21/port-forwarding-with-xinetd/