Entware/Entware #561: rmlint or rdfind

treeent, 2020-11-24:
Hi,

I’d like to use an efficient duplicate finder on my Entware-improved Synology NAS. I know “jdupes” is available but it is not efficient enough, esp. when dealing with intricate folder tree and a huge volume of data. In my experience, “rmlint” (or “rdfind”) offer way more versatility and performance. As each has (dis)advantages for specific situations, having them both would be wonderful. If only one, then let it be “rmlint” (a bit slower but with more fine tuning possible).

rmlint: https://github.com/sahib/rmlint
rdfind: https://github.com/pauldreik/rdfind

Thanks in advance!

- [v] No similar package already added and/or this one got some new killer feature(s),
- [v] Package is in active development (at least, not abandoned for 5 years or more),
- [v] I'm ready to test added package on my platform [e.g. mipselsf-k3.4] and get a feedback about it.

ryzhovau, 2021-04-04:
`rmlint` looks like a Python program with no any code compiled. Please, try to build it as any other Python module https://github.com/Entware/Entware/wiki/Self-installation-of-python-modules.

closed 2021-04-04 by ryzhovau

AdrianSosic, 2021-04-05:
Hi @ryzhovau! Thanks for your answer. Unfortunately, trying to install rmlint via pip does not work:

```
pip install rmlint
ERROR: Could not find a version that satisfies the requirement rmlint
ERROR: No matching distribution found for rmlint
```

AdrianSosic, 2021-04-11:
@ryzhovau: can we reopen this issue?

ryzhovau, 2021-04-12:
`rdfind` already added. And no plan to add `rmlint`, sorry.