Entware/Entware #563: incron 0.5.12 creates zombie children

token0, 2020-12-03:
incron 0.5.12 creates zombie children after a while, and eventually depletes system resources.

_Possible regression on RPM package from EPEL on EL7 #52
https://github.com/ar-/incron/issues/52_

_incrond forks a process on invocation which is never killed #65
https://github.com/ar-/incron/issues/65_

_Bug 1656939 - Incron causes defunct processes
https://bugzilla.redhat.com/show_bug.cgi?id=1656939_

_Bug 1666590 - incrond multiply processes
https://bugzilla.redhat.com/show_bug.cgi?id=1666590_

_Bug 25130 - incrond forks a process on invocation which is never killed
https://bugs.mageia.org/show_bug.cgi?id=25130_

_Debian Bug report logs - #930526 incron creates zombie processes
https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=930526_

```
# opkg --version
opkg version f2166a89b7e96e2c60002959731eebc5f45fa318 (2020-05-07)

# opkg list|grep incron
incron - 0.5.12-1 - This program is the "inotify cron" system.  You can use it a similar way as the regular cron. The difference is that the inotify cron handles filesystem events rather than time periods.

# incrond --version
incrond 0.5.12
```

**check.sh** is my script, runned by incrontab; let's see, how much zombies incron made:
```
# ps | grep check.sh | grep -v grep -c
300

```
And my system is up just for about 16 hours
```
# uptime
 11:06:09 up 15:54,  load average: 0.00, 0.00, 0.00
```

token0, 2020-12-03:
fast temporary fix is to downgrade to 0.5.10
```
wget https://bin.entware.net/mipselsf-k3.4/archive/incron_0.5.10-1_mipsel-3.4.ipk
opkg install incron_0.5.10-1_mipsel-3.4.ipk  --force-downgrade 
```

ryzhovau, 2020-12-04:
Debian guys got necessary patch which I should apply too.

Thanks for feedback!

closed 2021-04-06 by ryzhovau