Entware/Entware #564: Entware rsync v3.2.3 vs. v3.1.3 loss of iconv capability

gilgrissom, 2020-12-05:
**Describe the bug**
rsync: Apparently the recently rolled out upgrade to Entware rsync has changed/disabled a few functionalities and now iconv capability is lost.
Just wondering if this was an oversight and rsync package can be regenerated maintaining compatibility or other technical issues/choices prevented iconv to be enabled.

**Environment (please complete the following information):**
 - Entware feed: armv7sf-k3.2
 - Device Asus RT-AC3200
 - Firmware version Freshtomato 2020.3 K26ARM USB AIO-128K

https://bin.entware.net/armv7sf-k3.2/**rsync_3.2.3-2_armv7-3.2.ipk**
rsync  version 3.2.3  protocol version 31
Copyright (C) 1996-2020 by Andrew Tridgell, Wayne Davison, and others.
Web site: https://rsync.samba.org/
Capabilities:
    64-bit files, 64-bit inums, 32-bit timestamps, 64-bit long ints,
    no socketpairs, hardlinks, no hardlink-specials, symlinks, IPv6, atimes,
    batchfiles, inplace, append, ACLs, xattrs, optional protect-args,
    **_no iconv_**, symtimes, prealloc, stop-at, no crtimes
Optimizations:
    no SIMD, no asm, no openssl-crypto
Checksum list:
    md5 md4 none
Compress list:
    zlibx zlib none

https://bin.entware.net/armv7sf-k3.2/archive/**rsync_3.2.3-1_armv7-3.2.ipk**
rsync  version 3.1.3  protocol version 31
Copyright (C) 1996-2018 by Andrew Tridgell, Wayne Davison, and others.
Web site: http://rsync.samba.org/
Capabilities:
    64-bit files, 64-bit inums, 32-bit timestamps, 64-bit long ints,
    no socketpairs, hardlinks, symlinks, IPv6, batchfiles, inplace,
    append, ACLs, xattrs, **_iconv_**, symtimes, prealloc

zyxmon, 2020-12-05:
iconv was disabled for rsync in September 2020 (previous build) as in upstream - https://github.com/openwrt/packages/blob/master/net/rsync/Makefile#L49

closed 2020-12-05 by zyxmon

gilgrissom, 2021-01-02:
As soon as 3.2.3 is intercepted in https://bin.entware.net/armv7sf-k3.2/ build, I'll give it a try and report back.

gilgrissom, 2021-01-10:
Tested and apparently working with version:
http://bin.entware.net/armv7sf-k3.2/rsync_3.2.3-3_armv7-3.2.ipk
Thanks a ton!