Entware/Entware #565: NTS missing in Chrony 4.0

JGrana01, 2020-12-05:
**Describe the bug**
[chrony]: It appears chrony was built without NTS support. The chrony build configure script checks for GnuTLS. If not present, NTS support is not compiled. It doesn't appear that GnuTLS was in the compile/depends path?

**Environment (please complete the following information):**
 - Entware feed: [arm64]
 - Device [RT-AX88U]
 - Firmware version [asuswrt-merlin]

jackyaz, 2020-12-05:
See https://github.com/Entware/Entware/issues/552#issuecomment-739242532

butaford, 2020-12-11:
add to config `CONFIG_PACKAGE_chrony-nts=m`  in https://github.com/Entware/Entware/blob/2df8313741e25bf260a4a3bb1d893371b11c8d51/configs/aarch64-3.10.config#L5985 and compile

jackyaz, 2021-02-22:
Any updates on this making it into the repo, please?

ryzhovau, 2021-02-23:
Depends on free time we have. We'll sync with upstream on the next week.

jackyaz, 2021-02-23:
@ryzhovau Thanks!

jackyaz, 2021-03-23:
I can see package updates tonight, but nothing for chrony? :(

closed 2021-04-06 by ryzhovau