Entware/Entware #566: samba4-server doesn't start

Diskutant, 2020-12-05:
**Describe the bug**
[samba4-server]: I installed entware, samba4-admin, samba4-libs, samba4-server and samba4-utils on a ZyXEL NAS 542.
After reboot no smbd is started. So I tried it manually with "-d 9" option, it looks like it started, but it actually isn't.

The last few lines are: 

`[2020/12/03 17:23:51.950786,  4] ../../source3/param/loadparm.c:3974(lp_load_ex)
  pm_process() returned Yes
[2020/12/03 17:23:51.951006,  7] ../../source3/param/loadparm.c:4309(lp_servicenumber)
  lp_servicenumber: couldn't find homes
[2020/12/03 17:23:51.951250,  4] ../../source3/smbd/sec_ctx.c:174(get_current_groups)
  get_current_groups: user is in 1 groups: 0`

according to the following thread that "couldn't find homes" error is caused by wrong libnss_winbind*.so files.
http://samba.2283325.n4.nabble.com/RE-Samba-SaMBa-4-homedir-mapping-lp-servicenumber-td4643737.html

I'm not sure if that's really the cause of smbd just quitting, but as there is no other message that looks like an error this is my best guess.
So is it possible the libnss_winbind*.so files are indeed wrong?
Or any other ideas why smbd just quits immediately?

**Environment:**
 - Entware feed: armv7sf-k3.2/
 - Device: ZyXEL NAS 542
 - Firmware version: firmware V5.21(ABAG.6)C0

zyxmon, 2020-12-05:
I have just installed samba4 on a arm7 NAS. Samba4 starts correctly with
`/opt/etc/init.d/S91smb start` and a configured share is visible.
Cannot confirm the bug. Probably samba was not started correctly or there is some interference with firmware samba.

closed 2020-12-05 by zyxmon

zyxmon, 2020-12-05:
PS I have used a very simple smb.conf to share a single directory.

Diskutant, 2020-12-05:
> Probably samba was not started correctly 

yes, that's the issue. It just prints no error message, except for that "lp_servicenumber: couldn't find homes" message when using the -d 9 option.

> or there is some interference with firmware samba.

firmware samba isn't running, so that's not it.

`/opt/etc/init.d/S91smb start` just prints FAIL

`sudo /opt/sbin/smbd` and then `echo $?` prints 1

maybe on your system the correct  libnss_winbind*.so files are already installed?

    md5sum /opt/lib/libnss_win*.so
    9d9df271eb6f98aae978c1fe40a23c99  /opt/lib/libnss_winbind.so
    0dded0e04198456bff2da764d88caa99  /opt/lib/libnss_wins.so

is yours different?

> PS I have used a very simple smb.conf to share a single directory.

What would the most simple working smb.conf look like?

zyxmon, 2020-12-05:
md5sum is the same.

zyxmon, 2020-12-05:
Just append to /opt/etc/samba/smb.conf something similar to

```
[Entware]
       path = /opt
       guest ok = yes
       force user = root
       writable = yes
       comment = USB Flash Drive
```