Entware/Entware #567: Could you please add macchanger package

iacchi, 2020-12-07:
Hello,
thank you for the effort in maintaining this software repository. I would like to know if it would be possible to include the package macchanger inside the repository. This is the project homepage:

https://github.com/alobbs/macchanger

This package is needed to spoof the mac address of a device, which could be useful in certain circumstances (I know for sure that I need it). The project hasn't been updated in many years, but I think it's mostly because there's nothing to be done on it, rather than it being abandoned.

butaford, 2020-12-11:
Macchanger already exist https://github.com/Entware/entware-packages/tree/master/utils/macchanger
Add the /opt prefix to the Makefile and compile
[image]

iacchi, 2020-12-11:
I thought it could be made available in binary form within the opkg package manager. If compiling is the only way, I'll try to start figuring out how to do that. I've seen there's a page on the wiki that explains things.

butaford, 2020-12-11:
What architecture?

iacchi, 2020-12-11:
Intel x64

butaford, 2020-12-11:
try
[macchanger_1.7.0-3_x64-3.2.ipk.zip](https://github.com/Entware/Entware/files/5679142/macchanger_1.7.0-3_x64-3.2.ipk.zip)

iacchi, 2020-12-11:
Thank you very much! Really appreciated :)

ryzhovau, 2020-12-11:
> Intel x64

Unfortunately, Its useless for any other architecture. Every vendor got it's own way to deal with this things, so there's no any unified interface is available on embedded devices.

closed 2020-12-11 by ryzhovau