Entware/Entware #569: Question : Guide to create ssh server and authorisation keys for remote access

nodecentral, 2020-12-15:
Hi,

I’m looking to set my QNAP NAS up as an ssh server and have some people to connect (via cert/no pw prompt) to run the commands they need.  I’ve tried to find a step by step guide, and came across this - https://wiki.qnap.com/wiki/Replace_ssh_with_Qnapware_OpenSSH ? 

While it talks about ‘qnapware’ is the process the same for Entware ?

zyxmon, 2020-12-15:
Qnapware is obsolete now. In Qnapware - all files were installed in /Apps/opt, in Entware files are in /opt.
The Entware startup script is installed automatically in /opt/etc/init.d

closed 2020-12-15 by zyxmon

nodecentral, 2020-12-15:
Sorry @zyxmon - Is there a guide I can follow to achieve the same thing with Entware ?