Entware/Entware #572: uboot-envtools: package support

butaford, 2020-12-16:
Binaries for tests:
[uboot-envtools_2020.04-1_armv7-2.6.zip](https://github.com/Entware/Entware/files/5701879/uboot-envtools_2020.04-1_armv7-2.6.zip)
[uboot-envtools_2020.04-1_armv7-3.2.zip](https://github.com/Entware/Entware/files/5701880/uboot-envtools_2020.04-1_armv7-3.2.zip)
[uboot-envtools_2020.04-1_mips-3.4.zip](https://github.com/Entware/Entware/files/5701881/uboot-envtools_2020.04-1_mips-3.4.zip)
[uboot-envtools_2020.04-1_mipsel-3.4.zip](https://github.com/Entware/Entware/files/5701882/uboot-envtools_2020.04-1_mipsel-3.4.zip)
[uboot-envtools_2020.04-1_x64-3.2.zip](https://github.com/Entware/Entware/files/5701883/uboot-envtools_2020.04-1_x64-3.2.zip)
[uboot-envtools_2020.04-1_x86-2.6.zip](https://github.com/Entware/Entware/files/5701884/uboot-envtools_2020.04-1_x86-2.6.zip)
[uboot-envtools_2020.04-1_aarch64-3.10.zip](https://github.com/Entware/Entware/files/5701885/uboot-envtools_2020.04-1_aarch64-3.10.zip)
[uboot-envtools_2020.04-1_armv5-3.2.zip](https://github.com/Entware/Entware/files/5701886/uboot-envtools_2020.04-1_armv5-3.2.zip)

Source codes:
[uboot-envtools_src.zip](https://github.com/Entware/Entware/files/5701888/uboot-envtools_src.zip)

butaford, 2021-02-26:
не забудьте добавить :-)

ryzhovau, 2021-04-06:
Посеба!

closed 2021-04-06 by ryzhovau