Entware/Entware #573: update unbound to version 1.13

laracroftonline, 2020-12-16:
Could you update the unbound package from unbound to 1.13

butaford, 2020-12-17:
[unbound_1.13.0-1_aarch64-3.10.zip](https://github.com/Entware/Entware/files/5707136/unbound_1.13.0-1_aarch64-3.10.zip)
[unbound_1.13.0-1_armv5-3.2.zip](https://github.com/Entware/Entware/files/5707137/unbound_1.13.0-1_armv5-3.2.zip)
[unbound_1.13.0-1_armv7-2.6.zip](https://github.com/Entware/Entware/files/5707138/unbound_1.13.0-1_armv7-2.6.zip)
[unbound_1.13.0-1_armv7-3.2.zip](https://github.com/Entware/Entware/files/5707139/unbound_1.13.0-1_armv7-3.2.zip)
[unbound_1.13.0-1_mips-3.4.zip](https://github.com/Entware/Entware/files/5707140/unbound_1.13.0-1_mips-3.4.zip)
[unbound_1.13.0-1_mipsel-3.4.zip](https://github.com/Entware/Entware/files/5707141/unbound_1.13.0-1_mipsel-3.4.zip)
[unbound_1.13.0-1_x64-3.2.zip](https://github.com/Entware/Entware/files/5707142/unbound_1.13.0-1_x64-3.2.zip)
[unbound_1.13.0-1_x86-2.6.zip](https://github.com/Entware/Entware/files/5707143/unbound_1.13.0-1_x86-2.6.zip)

laracroftonline, 2020-12-17:
> [unbound_1.13.0-1_aarch64-3.10.zip](https://github.com/Entware/Entware/files/5707136/unbound_1.13.0-1_aarch64-3.10.zip) [...]

Thanks still learning github

ryzhovau, 2020-12-17:
@laracroftonline, this package came from upstream with no any changes. Please, [ask](https://github.com/openwrt/packages/issues) OpenWrt guys to update it. We sync with them periodically, so new version will arrive to Entware then.

@butaford, thanks for cooperation. This way is really useful to prove new package is working, but can't be the main solution for our repo. The best way is to make PR against original code. We'll gladly accept PRs for our own [golang](https://github.com/Entware/entware-go), [rtndev](https://github.com/Entware/rtndev) or [oldports](https://github.com/Entware/entware-oldpackages-ports) feeds, otherwise, make PRs for upstream. Any new compiled *.ipk files will be deleted from @Entware Issues and I hope you understand and share our vision.

ryzhovau, 2021-03-25:
Done.

closed 2021-03-25 by ryzhovau