Entware/Entware #574: Nginx ERR_HTTP2_COMPRESSION_ERROR

jumpsmm7, 2020-12-17:
Package-
nginx version: nginx/1.18.0
built by gcc 8.4.0 (OpenWrt GCC 8.4.0 r1372-dac52b08)
built with OpenSSL 1.1.1h  22 Sep 2020
TLS SNI support enabled
configure arguments: --crossbuild=Linux::aarch64 --with-cc=aarch64-openwrt-linux-gnu-gcc --with-cc-opt='-I/media/ware3/Entware.2020.11/staging_dir/target-aarch64_cortex-a53_glibc-2.27/opt/include -I/media/ware3/Entware.2020.11/staging_dir/toolchain-aarch64_cortex-a53_gcc-8.4.0_glibc-2.27/include -O2 -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -I/media/ware3/Entware.2020.11/staging_dir/target-aarch64_cortex-a53_glibc-2.27/opt/include/libxml2 -Wno-error=parentheses -Wno-error=implicit-fallthrough -Wno-error=cast-function-type' --with-ld-opt='-L/media/ware3/Entware.2020.11/staging_dir/target-aarch64_cortex-a53_glibc-2.27/opt/lib -Wl,-rpath,/opt/lib -Wl,-rpath-link=/media/ware3/Entware.2020.11/staging_dir/target-aarch64_cortex-a53_glibc-2.27/opt/lib -Wl,--dynamic-linker=/opt/lib/ld-linux-aarch64.so.1 -L/media/ware3/Entware.2020.11/staging_dir/toolchain-aarch64_cortex-a53_gcc-8.4.0_glibc-2.27/lib -lxml2' --prefix=/opt --conf-path=/opt/etc/nginx/nginx.conf --http-log-path=/opt/var/log/nginx/access.log --error-log-path=/opt/var/log/nginx/error.log --lock-path=/opt/var/lock/nginx.lock --pid-path=/opt/var/run/nginx.pid --modules-path=/opt/lib/nginx --http-client-body-temp-path=/opt/var/lib/nginx/body --http-fastcgi-temp-path=/opt/var/lib/nginx/fastcgi --http-proxy-temp-path=/opt/var/lib/nginx/proxy --http-scgi-temp-path=/opt/var/lib/nginx/scgi --http-uwsgi-temp-path=/opt/var/lib/nginx/uwsgi --with-debug --with-pcre-jit --with-http_ssl_module --with-http_stub_status_module --with-http_realip_module --with-http_auth_request_module --with-http_v2_module --with-http_dav_module --with-http_slice_module --without-http_upstream_zone_module --without-stream_upstream_zone_module --with-http_addition_module --with-http_flv_module --with-http_gunzip_module --with-http_gzip_static_module --with-http_image_filter_module=dynamic --with-http_mp4_module --with-http_random_index_module --with-http_secure_link_module --with-http_sub_module --with-http_xslt_module=dynamic --with-mail=dynamic --with-mail_ssl_module --with-stream=dynamic --with-stream_ssl_module --add-dynamic-module=/media/ware3/Entware.2020.11/build_dir/target-aarch64_cortex-a53_glibc-2.27/nginx-1.18.0/modules/http-headers-more-filter --add-dynamic-module=/media/ware3/Entware.2020.11/build_dir/target-aarch64_cortex-a53_glibc-2.27/nginx-1.18.0/modules/http-auth-pam --add-dynamic-module=/media/ware3/Entware.2020.11/build_dir/target-aarch64_cortex-a53_glibc-2.27/nginx-1.18.0/modules/http-cache-purge --add-dynamic-module=/media/ware3/Entware.2020.11/build_dir/target-aarch64_cortex-a53_glibc-2.27/nginx-1.18.0/modules/http-dav-ext --add-dynamic-module=/media/ware3/Entware.2020.11/build_dir/target-aarch64_cortex-a53_glibc-2.27/nginx-1.18.0/modules/http-ndk --add-dynamic-module=/media/ware3/Entware.2020.11/build_dir/target-aarch64_cortex-a53_glibc-2.27/nginx-1.18.0/modules/http-echo --add-dynamic-module=/media/ware3/Entware.2020.11/build_dir/target-aarch64_cortex-a53_glibc-2.27/nginx-1.18.0/modules/http-fancyindex --add-dynamic-module=/media/ware3/Entware.2020.11/build_dir/target-aarch64_cortex-a53_glibc-2.27/nginx-1.18.0/modules/nchan --add-dynamic-module=/media/ware3/Entware.2020.11/build_dir/target-aarch64_cortex-a53_glibc-2.27/nginx-1.18.0/modules/http-lua --add-dynamic-module=/media/ware3/Entware.2020.11/build_dir/target-aarch64_cortex-a53_glibc-2.27/nginx-1.18.0/modules/rtmp --add-dynamic-module=/media/ware3/Entware.2020.11/build_dir/target-aarch64_cortex-a53_glibc-2.27/nginx-1.18.0/modules/http-uploadprogress --add-dynamic-module=/media/ware3/Entware.2020.11/build_dir/target-aarch64_cortex-a53_glibc-2.27/nginx-1.18.0/modules/http-upstream-fair --add-dynamic-module=/media/ware3/Entware.2020.11/build_dir/target-aarch64_cortex-a53_glibc-2.27/nginx-1.18.0/modules/http-subs-filter
Entware Feed- aarch64-k3.10/nginx-extras_1.18.0-2_aarch64-3.10.ipk
Device-Asus RT-AX88U router
Firmware version- RMerlin 386.1

**BUG DETAILS** 
It appears using http2 gives a compression error in "some" browsers, Chrome and Microsoft Edge. The issue does not exist if Firefox. 
This is what the error says in Chrome and Microsoft Edge. 
The webpage at  ******my domains address******** might be temporarily down or it may have moved permanently to a new web address.
ERR_HTTP2_COMPRESSION_ERROR
It looks like this issue was once reported before and closed. It may not have been given proper scrutiny, but it appears I am also affected by this strange issue. I have tried multiple devices receiving the same results.

jumpsmm7, 2020-12-19:
This is a continued issue for the this package of NGINX, I switched to NGINX-SSL package and the issue went away. I am closing this since I found a happy medium.

barnumbirr, 2021-09-18:
Issue still persists in all (Firefox, Chrome and Edge) with the same `ERR_HTTP2_COMPRESSION_ERROR` error.

```bash
$ nginx -V
nginx version: nginx/1.18.0
built by gcc 8.4.0 (OpenWrt GCC 8.4.0 r1653-fe90b72b)
built with OpenSSL 1.1.1k  25 Mar 2021
TLS SNI support enabled
configure arguments: --crossbuild=Linux::x86_64 --with-cc=x86_64-openwrt-linux-gnu-gcc --with-cc-opt='-I/build/me/E.x64/staging_dir/toolchain-x86_64_gcc-8.4.0_glibc-2.27/include -I/build/me/E.x64/staging_dir/target-x86_64_glibc-2.27/opt/lib/libiconv-full/include -I/build/me/E.x64/staging_dir/target-x86_64_glibc-2.27/opt/lib/libintl-full/include -O2 -pipe -fomit-frame-pointer -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -I/build/me/E.x64/staging_dir/target-x86_64_glibc-2.27/opt/lib/libiconv-full/include -I/build/me/E.x64/staging_dir/target-x86_64_glibc-2.27/opt/lib/libintl-full/include -I/build/me/E.x64/staging_dir/target-x86_64_glibc-2.27/opt/include/libxml2 -Wno-error=parentheses -Wno-error=implicit-fallthrough -Wno-error=cast-function-type' --with-ld-opt='-Wl,--dynamic-linker=/opt/lib/ld-linux-x86-64.so.2 -Wl,-rpath=/opt/lib -L/build/me/E.x64/staging_dir/toolchain-x86_64_gcc-8.4.0_glibc-2.27/lib -L/build/me/E.x64/staging_dir/target-x86_64_glibc-2.27/opt/lib/libiconv-full/lib -Wl,-rpath-link=/build/me/E.x64/staging_dir/target-x86_64_glibc-2.27/opt/lib/libiconv-full/lib -L/build/me/E.x64/staging_dir/target-x86_64_glibc-2.27/opt/lib/libintl-full/lib -Wl,-rpath-link=/build/me/E.x64/staging_dir/target-x86_64_glibc-2.27/opt/lib/libintl-full/lib -lxml2' --prefix=/opt --conf-path=/opt/etc/nginx/nginx.conf --http-log-path=/opt/var/log/nginx/access.log --error-log-path=/opt/var/log/nginx/error.log --lock-path=/opt/var/lock/nginx.lock --pid-path=/opt/var/run/nginx.pid --modules-path=/opt/lib/nginx --http-client-body-temp-path=/opt/var/lib/nginx/body --http-fastcgi-temp-path=/opt/var/lib/nginx/fastcgi --http-proxy-temp-path=/opt/var/lib/nginx/proxy --http-scgi-temp-path=/opt/var/lib/nginx/scgi --http-uwsgi-temp-path=/opt/var/lib/nginx/uwsgi --with-debug --with-pcre-jit --with-http_ssl_module --with-http_stub_status_module --with-http_realip_module --with-http_auth_request_module --with-http_v2_module --with-http_dav_module --with-http_slice_module --without-http_upstream_zone_module --without-stream_upstream_zone_module --with-http_addition_module --with-http_flv_module --with-http_gunzip_module --with-http_gzip_static_module --with-http_image_filter_module=dynamic --with-http_mp4_module --with-http_random_index_module --with-http_secure_link_module --with-http_sub_module --with-http_xslt_module=dynamic --with-mail=dynamic --with-mail_ssl_module --with-stream=dynamic --with-stream_ssl_module --add-dynamic-module=/build/me/E.x64/build_dir/target-x86_64_glibc-2.27/nginx-1.18.0/modules/http-headers-more-filter --add-dynamic-module=/build/me/E.x64/build_dir/target-x86_64_glibc-2.27/nginx-1.18.0/modules/http-auth-pam --add-dynamic-module=/build/me/E.x64/build_dir/target-x86_64_glibc-2.27/nginx-1.18.0/modules/http-cache-purge --add-dynamic-module=/build/me/E.x64/build_dir/target-x86_64_glibc-2.27/nginx-1.18.0/modules/http-dav-ext --add-dynamic-module=/build/me/E.x64/build_dir/target-x86_64_glibc-2.27/nginx-1.18.0/modules/http-ndk --add-dynamic-module=/build/me/E.x64/build_dir/target-x86_64_glibc-2.27/nginx-1.18.0/modules/http-echo --add-dynamic-module=/build/me/E.x64/build_dir/target-x86_64_glibc-2.27/nginx-1.18.0/modules/http-fancyindex --add-dynamic-module=/build/me/E.x64/build_dir/target-x86_64_glibc-2.27/nginx-1.18.0/modules/nchan --add-dynamic-module=/build/me/E.x64/build_dir/target-x86_64_glibc-2.27/nginx-1.18.0/modules/http-lua --add-dynamic-module=/build/me/E.x64/build_dir/target-x86_64_glibc-2.27/nginx-1.18.0/modules/rtmp --add-dynamic-module=/build/me/E.x64/build_dir/target-x86_64_glibc-2.27/nginx-1.18.0/modules/http-uploadprogress --add-dynamic-module=/build/me/E.x64/build_dir/target-x86_64_glibc-2.27/nginx-1.18.0/modules/http-upstream-fair --add-dynamic-module=/build/me/E.x64/build_dir/target-x86_64_glibc-2.27/nginx-1.18.0/modules/http-subs-filter
```

jumpsmm7, 2021-11-19:
Yes I just switched back to NGINX-EXTRA, and the compression error still exist. I temporarily used SSL, but switched back since there are options available with NGINX-EXTRA that I would like to use.

ryzhovau, 2021-12-09:
`nginx-extras` built exactly as in Debian and `nginx(-ssl)` came from OpenWrt with no any changes. Which one should I check?

jumpsmm7, 2021-12-15:
> nginx-extras

nginx-extras

ryzhovau, 2023-05-23:
I'm closing this for now as long as I've got nothing to do here.

closed 2023-05-23 by ryzhovau