Entware/Entware #577: Installing Entware on Asus RT-AC57U V3

zssasa7, 2020-12-21:
Hi,

Im'm trying to install Entware on my new Asus RT-AC57U V3 router based on this guide:
https://github.com/Entware/Entware-ng/wiki/Install-on-Asus-stock-firmware
I'm on the latest firmware (3.0.0.4.386_2164
As there is no custom firmware for this model, I'm doing it on stock firmware and getting this error after downloaded the package:
2020-12-21 08:37:17 (24.5 MB/s) - written to stdout [1359/1359]

Info: Creating folders...
Info: Deploying opkg package manager...
Downloading /opt/bin/opkg... success!
Downloading /opt/etc/opkg.conf... success!
Downloading /opt/etc/profile... success!
Downloading /opt/etc/init.d/rc.func... success!
Downloading /opt/etc/init.d/rc.unslung... success!
Info: Basic packages installation...
/opt/bin/opkg: line 1: syntax error: unexpected "("
/opt/bin/opkg: line 1: syntax error: unexpected "("

Do you have any idea whats wrong? Or maybe its just that entware can't be installed on my model?

Thanks in advance!

ryzhovau, 2020-12-21:
Please, provide `cat /proc/cpuinfo` dump here.

zssasa7, 2020-12-21:
admin@RT-AC57U_V3-91C0:/tmp/home/root# cat /proc/cpuinfo
system type             : Qualcomm Atheros QCA5502 rev 0
machine                 : Qualcomm Atheros APJET01 reference board
processor               : 0
cpu model               : MIPS 74Kc V5.0
BogoMIPS                : 385.84
wait instruction        : yes
microsecond timers      : yes
tlb_entries             : 32
extra interrupt vector  : yes
hardware watchpoint     : yes, count: 4, address/irw mask: [0x0000, 0x0ff8, 0x0ff8, 0x0ff8]
ASEs implemented        : mips16 dsp
shadow register sets    : 1
kscratch registers      : 0
core                    : 0
VCED exceptions         : not available
VCEI exceptions         : not available

ryzhovau, 2020-12-22:
* Entware-ng is EoS for now, sorry.
* This is MIPS router, so the only right installer for this model is http://pkg.entware.net/binaries/mipsel/installer/installer.sh If it fails, no other options left, sorry.

closed 2020-12-22 by ryzhovau