Entware/Entware #580: mail/alpine missing package, but Makefile exists

thosoo, 2021-01-02:
**Describe the bug**
[mail/alpine]: I tried to install this package after finding it in the package [sources](https://github.com/Entware/entware-packages/tree/master/mail/alpine). However, on installation I just get the following:
```
DD-WRT# opkg install alpine
Unknown package 'alpine'.
Collected errors:
 * opkg_install_cmd: Cannot install package alpine.
```
After trying to find the `.ipk` [here](https://pkg.entware.net/binaries/armv7/), the package does not exist. There seems to be something broken.

**Environment (please complete the following information):**
 - Entware feed: armv7l
 - Device: Linksys EA6400
 - Firmware version: DD-WRT v3.0-r45229 std (01/01/21)

zyxmon, 2021-01-03:
Entware-ng is outdated and is not supported, sorry.

closed 2021-01-03 by zyxmon

thosoo, 2021-01-03:
I dont think I am using -ng. I have just pasted the wrong repository link.. 
At http://bin.entware.net/armv7sf-k3.2/ can also not find this package.. 
My opkg.conf states the following 
```
1 src/gz entware http://bin.entware.net/armv7sf-k3.2
2 dest root /
3 dest ram /opt/tmp
4 lists_dir ext /opt/var/opkg-lists
5 option tmp_dir /opt/tmp
6 arch all 100
7 arch armv7-3.2 160
```

zyxmon, 2021-01-04:
All packages for this arch - http://bin.entware.net/armv7sf-k3.2/Packages.html
Not all packages are ported from openwrt to entware.