Entware/Entware #582: Python3 (3.9.0-2) install failed if python (2) is installed

father-mande, 2021-01-03:
Python3 install
error message : 
opkg install python3
Installing python3 (3.9.0-2) to root...
Downloading http://bin.entware.net/x64-k3.2/python3_3.9.0-2_x64-3.2.ipk
Installing python3-base (3.9.0-2) to root...
Downloading http://bin.entware.net/x64-k3.2/python3-base_3.9.0-2_x64-3.2.ipk
Collected errors:
 * check_data_file_clashes: Package python3-base wants to install file /opt/bin/python
        But that file is already provided by package  * python-base
 * opkg_install_cmd: Cannot install package python3.

Philippe.

zyxmon, 2021-01-03:
This is a known issue. You need to remove python2 before installing python3.

closed 2021-01-03 by zyxmon

father-mande, 2021-01-03:
Hi,

after tests removing /opt/bin/python is enough  ... so python2.7 can be called but using it's full name python2.7

Philippe.

zyxmon, 2021-01-03:
Removing should be done with `opkg remove ....` probably some force flags may be needed.
You can also install with --force-overwrite flag