Entware/Entware #586: Incorrect sha256sum for some packages on armv7sf-k3.2

RikshaDriver, 2021-01-13:
When trying to install packages on QNAP hardware running armv7sf-k3.2, a sha256 mismatch error is observed for a number of packages. For example, git, libmagic, gcc all report similar mistmatches:

Installing gcc (7.4.0-5) to root...
Downloading http://bin.entware.net/armv7sf-k3.2/gcc_7.4.0-5_armv7-3.2.ipk
Collected errors:
 * opkg_install_pkg: Package gcc sha256sum mismatch. Either the opkg or the package index are corrupt. Try 'opkg update'.
 * opkg_install_cmd: Cannot install package gcc.

In the case of gcc, I've compared the sha256 checksum found in:
http://bin.entware.net/armv7sf-k3.2/Packages.gz

and the actual ipk download. There is definitely a mismatch.

Seems a number of packages need checksums to be recomputed.

ryzhovau, 2021-01-13:
Please, do `opkg update` and check again. It should be fine for now.

RikshaDriver, 2021-01-13:
Gave it another go. Still getting the checksum mismatches.

zyxmon, 2021-01-13:
Cannot confirm it
```
/share/CACHEDEV1_DATA/.qpkg/Entware/root # opkg update
Downloading http://bin.entware.net/armv7sf-k3.2/Packages.gz
Updated list of available packages in /opt/var/opkg-lists/entware
/share/CACHEDEV1_DATA/.qpkg/Entware/root # opkg install gcc
Installing gcc (7.4.0-5) to root...
Downloading http://bin.entware.net/armv7sf-k3.2/gcc_7.4.0-5_armv7-3.2.ipk
Installing libbfd (2.34-1) to root...
Downloading http://bin.entware.net/armv7sf-k3.2/libbfd_2.34-1_armv7-3.2.ipk
Installing libopcodes (2.34-1) to root...
Downloading http://bin.entware.net/armv7sf-k3.2/libopcodes_2.34-1_armv7-3.2.ipk
Installing libctf (2.34-1) to root...
Downloading http://bin.entware.net/armv7sf-k3.2/libctf_2.34-1_armv7-3.2.ipk
Installing objdump (2.34-1) to root...
Downloading http://bin.entware.net/armv7sf-k3.2/objdump_2.34-1_armv7-3.2.ipk
Installing ar (2.34-1) to root...
Downloading http://bin.entware.net/armv7sf-k3.2/ar_2.34-1_armv7-3.2.ipk
Installing binutils (2.34-1) to root...
Downloading http://bin.entware.net/armv7sf-k3.2/binutils_2.34-1_armv7-3.2.ipk
Configuring libbfd.
Configuring libopcodes.
Configuring libctf.
Configuring objdump.
Configuring ar.
Configuring binutils.
Configuring gcc.
There are no *-dev packages in Entware(with few exceptions)!
Please install headers as described in the wiki:
https://github.com/Entware/Entware/wiki
/share/CACHEDEV1_DATA/.qpkg/Entware/root # cat /etc/version_info
commit 769c72ed5766fb32b426ef6de8440e25980081ef
Date:   2020-09-28 17:13:43 +0800
================================================
* QTS_4.3.5
  remotes/origin/QTS_4.3.5
/share/CACHEDEV1_DATA/.qpkg/Entware/root #

```

zyxmon, 2021-01-13:
@RikshaDriver - may be you have bad internet connection!
1. Corrupted downloads
or
2. Internet provider caches some files (was a problem in some cases in China)

butaford, 2021-01-13:
@RikshaDriver 
[image]

RikshaDriver, 2021-01-13:
Interesting. If I wget ipk via http, the sha256 checksum is:
[~] # sha256sum gcc_7.4.0-5_armv7-3.2.ipk
3c7b2174588a273fc25db606250db1e10d731e0b7a3fcad02498c022a3043c3e  gcc_7.4.0-5_armv7-3.2.ipk

If I wget via https, then sha256sum is:
[~] # sha256sum gcc_7.4.0-5_armv7-3.2.ipk.1
e1e85f1653bdbdd12a7fa30cd973fc458549a3a80c9dc24f3f5a5ce8d18baf97  gcc_7.4.0-5_armv7-3.2.ipk.1

files "appear" to be identical sizes:

36359407 Dec  1 06:20 gcc_7.4.0-5_armv7-3.2.ipk
36359407 Dec  1 06:20 gcc_7.4.0-5_armv7-3.2.ipk.1

What's going on here?

ryzhovau, 2021-01-14:
> What's going on here?

Some kind of MITM issue, may be it's up to Cloudflare which we're currently using.

ryzhovau, 2021-04-06:
Closing for now. Other armv7 users can't reproduce that.

closed 2021-04-06 by ryzhovau