Entware/Entware #587: Opkg "Illegal Instruction" with Motorola MD1600

SLGShark6, 2021-01-17:
**Describe the bug**
opkg returns "Illegal Instruction" on every call. Tried installing onto USB flash drive formatted with ext2/3/4 filesystems and mounted to "/opt", using the generic and alternative (listed [Here](https://github.com/Entware/Entware/wiki/Alternative-install-vs-standard)) for "mips (kernel >= 3.4)", same issue all around. Tried "mipsel (kernel >= 3.4)" and a few others but no dice. Cant find much information on the CPU beside this [Wiki](https://wikidevi.wi-cat.ru/BMIPS4350). Not sure what I'm doing wrong.

**Environment (please complete the following information):**
 - Entware feed: mipssf-k3.4
 - Device: Motorola MD1600 DSL Cable Modem/ Router
 - CPU Model: Broadcom BMIPS4350 v8.0
 - System Type: 963167GWV_004R
 - Linux Version 3.4.11-rt19
 - Busybox Version: 1.17.2

ryzhovau, 2021-04-06:
Can't reproduce that, we've got no any MD1600 or similar devices.

closed 2021-04-06 by ryzhovau

SLGShark6, 2021-04-07:
Any suggestions on what may cause that issue? any way I can assist to six this issue then?

zyxmon, 2021-04-07:
The most common case is using mips installer/repo for mipsel device or vice a versa. I do not think it is yours case,