Entware/Entware #588: ZeroTier doesn't work

glblduh, 2021-01-18:
**Describe the bug**
ZeroTier: A clear and concise description of what the bug is and how to reproduce it. 

**When i join my zerotier network the it successfully joins the network and is seen as online on the zerotier central, but when other peers tries to ping or access the ssh into it, it doesn't connect.**

**To reproduce it I just installed zerotier from the repo then join my network then used my laptop to ping it.**

**Environment (please complete the following information):**
 - Entware feed: [e.g. mipselsf-k3.4]
 - **armv7sf-k3.2**
 - Device [e.g. Asus RT-AX88U router]
 - **My Cloud Home**
 - Firmware version [e.g. asuswrt-merlin]
 - **Android**

zyxmon, 2021-01-18:
Entware developers do noy use zerotier. Try to contact other zerotier users. We have reports that zerotier is working in Entware. One needs to alow access to device through firewall.

glblduh, 2021-01-19:
Does Android have any firewall built-in that I can configure to allow access?

ryzhovau, 2021-01-19:
I'm very sorry, we do not provide personal support here. Please, ask WD community for assistance.

glblduh, 2021-01-19:
Okay, thanks!

closed 2021-01-19 by glblduh