Entware/Entware #589: CrowdSec

DougNix, 2021-01-18:
**Describe the solution you'd like**
A clear and concise description of new features. Provide project URL and any other helpful context.

I recently learned about an open-source, crowd-based security application called CrowdSec: https://crowdsec.net/. The developers have confirmed that the package will run on ARM processors, and I think that offering a means to deploy this software on home-grade routers could be very helpful. From the website:
"CrowdSec is a free & open-source security automation tool leveraging local IP behavior detection & a community-powered IP reputation system."

- [X] No similar package already added and/or this one got some new killer feature(s),
- [X Package is in active development (at least, not abandoned for 5 years or more),
- [X] I'm ready to test added package on my platform [e.g. mipselsf-k3.4] and get a feedback about it.

ryzhovau, 2021-04-09:
Looks promising, still, can be too heavy for some embedded devices. We've managed to compile it's binaries, but have no idea how to configure it. We'll gladly add it to repo if it will be proven as working. You didn't tell about your current feed, so it's a mipsel-only:)
```
opkg install http://bin.entware.net/mipselsf-k3.4/test/crowdsec_1.0.12-1_mipsel-3.4.ipk
```

DougNix, 2021-04-10:
Hi Alexander,

I don’t  know how to configure it either. I was trying to get info from the devs about that. If I learn something I’ll let you know. 

Doug

closed 2021-04-10 by ryzhovau