Entware/Entware #590: dbus missing .conf files

HeliosFL, 2021-01-19:
dbus (1.9.20-1)

Unable to start dbus on a fresh installation 

Error message: Failed to start message bus: Failed to open "/opt/share/dbus-1/system.conf": No such file or directory

Tried manually creating "/opt/share/dbus-1/" and --force-reinstall with no luck (files still missing). opkg info (see below) references .conf files under "/opt/etc/dbus-1/" however dbus-daemon is looking on "/opt/share/dbus-1/system.conf"

Entware feed: arm v7
Device: Netgear R7000
Firmware version: DD-WRT v3.0-r45454 std (01/15/21)

-------------------------------------------------------------------------------------------
Package: dbus
Version: 1.9.20-1
Depends: libc, libexpat, libdbus
Status: install user installed
Section: utils
Architecture: armv7soft
Maintainer: Steven Barth <cyrus@openwrt.org>
MD5Sum: b675c95c590b8b408d15958e64f87f24
Size: 93883
Filename: dbus_1.9.20-1_armv7soft.ipk
Conffiles:
 /opt/etc/dbus-1/session.conf 4e72df80ffd586fcc1f34c71c8cb2aeb94
 /opt/etc/dbus-1/system.conf a24fe118a2d9f4b312f6c8e91e55be6fe9
Source: feeds/packages/utils/dbus
Description: Simple interprocess messaging system (daemon)
Installed-Time: 1611026439

zyxmon, 2021-01-19:
You are using old Entware-ng that is not supported now. Please switch to new Entware. Setup scripts are here - https://github.com/Entware/Entware/wiki/Alternative-install-vs-standard

closed 2021-01-19 by zyxmon

HeliosFL, 2021-01-19:
Thanks!!! will switch right away.