Entware/Entware #591: Wireshark installs, but unable to launch

JimmieJoeJr, 2021-01-19:
**Wireshark installs, but says command unknown**
[Wireshark]: After installing wireshark, enter wireshark as command but erros saying wireshark is not a command

**DD-WRT Environment 
 - Entware feed: armv7-k3.2
 - Linksys EA6300v1 
 - Firmware version DD-WRT v3.0-r44715

zyxmon, 2021-01-19:
```
 opkg files wireshark | grep /opt/bin
/opt/bin/mergecap
/opt/bin/reordercap
/opt/bin/dumpcap
/opt/bin/idl2wrs
/opt/bin/sharkd
/opt/bin/rawshark
/opt/bin/capinfos
/opt/bin/captype
/opt/bin/text2pcap
/opt/bin/editcap
/opt/bin/tshark
/opt/bin/randpkt

```

closed 2021-01-19 by zyxmon

JimmieJoeJr, 2021-01-19:
I got the same file list that you showed but wireshark still does not launch.

After enter wireshark as a command:
-sh:  wireshark: not found

Sent from Joe's iPhone

zyxmon, 2021-01-20:
There is no wireshark binary, only a set of command line utilities.

JimmieJoeJr, 2021-01-20:
Thank you. It would be nice if there were this type of information about each of the packages somewhere. The description in the package list is very limited.

Sent from Joe's iPhone