Entware/Entware #595: duplicity

hdeller, 2021-01-26:
**Describe the solution you'd like**
Can you please provide a package for duplicity(Backup tool)?
http://duplicity.nongnu.org/
https://launchpad.net/duplicity/0.8-series/0.8.18

- [ ] No similar package already added and/or this one got some new killer feature(s),
didn't found one...

- [ ] Package is in active development (at least, not abandoned for 5 years or more),
Yes.

- [ ] I'm ready to test added package on my platform [e.g. mipselsf-k3.4] and get a feedback about it.
Yes, Synology DS920+ on x86_64.

zyxmon, 2021-01-26:
Please try to build it youself. It is a python prog - https://github.com/Entware/Entware/wiki/Self-installation-of-python-modules

hdeller, 2021-01-26:
zyxmon, thanks for the hint.
Sadly installing duplicity doesn't work out of the box.
I followed https://github.com/Entware/Entware/wiki/Self-installation-of-python-modules and ran
 pip3 install --upgrade pip setuptools
Afterward I downloaded duplicity: https://launchpad.net/duplicity/0.8-series/0.8.18/+download/duplicity-0.8.18.tar.gz
The setup.py script there then stops with

sh: msgfmt: command not found
Traceback (most recent call last):
  File "/volume1/homes/deller/duplicity-0.8.18/./setup.py", line 322, in <module>
    data_files=get_data_files(),
  File "/volume1/homes/deller/duplicity-0.8.18/./setup.py", line 123, in get_data_files
    assert not os.system(u"msgfmt po/%s.po -o po/%s/duplicity.mo" % (lang, lang)), lang

It seems the gettext package (for msgfmt) isn't available in entware.
Any ideas?
Helge

hdeller, 2021-01-26:
In the meantime I switched over to use Hyperbackup via rsync.
Nevertheless, a duplicity package would be nice in entware.

fhfournier, 2021-03-10:
Got that same probleme using with **virutalenv** i dont even understand why. I tried on different server with python2 and python3.

Same error using **pip install duplicity**

ryzhovau, 2021-04-06:
Should be heavily reworked, can't add it in present state: too much dependences I can't provide.

DavidKoot, 2021-06-05:
Will this ever be fixed, or is there any alternative for installing from tar? I came here, because I want to install Duplicity 0.8 on a Raspberry Pi. Installing through apt-get installs only 0.7 (not useful in my case). Adding the ppa does not work, since there is no distribution template for Raspbian/buster. Then my last resort is the tar file, and that gets stuck on the error mention by the OP.

VA1DER, 2021-12-14:
> It seems the gettext package (for msgfmt) isn't available in entware. Any ideas? Helge

I myself recently ran into this problem.  The entware and/or openwrt maintainers are quite adamantly uninterested in including gettext.  But with some work you can build it yourself and get the msgfmt utility you need.

1. Set up gcc and your dev environment as per [these instructions](https://github.com/Entware/Entware/wiki/Using-GCC-for-native-compilation).  I would install all minimal, basic, and recommended, autotools, and you definitely need the dev headers for your platform.  It's actually not all that much.  I didn't have to mess around with the environment variables too much... the script they provided /opt/bin/gcc_env.sh had pretty reasonable defaults.  Every time before you build anything make sure that you source that script:
`# . /opt/bin/gcc_env.sh`
The period-space before the script is important.  I added the above to my .profile so it's run automatically every time I log in, but depending on your device and how it's configured and how you've customized it, this may or may not be an easy prospect.  Just make sure that script is sourced before you do anything with your dev environment.
2. Get the [gettext source](https://ftp.gnu.org/pub/gnu/gettext/gettext-0.21.tar.gz), untar and cd into it:
`# wget https://ftp.gnu.org/pub/gnu/gettext/gettext-0.21.tar.gz`
`# /bin/tar -xvzf gettext-0.21.tar.gz`
`# cd gettext-0.21`
3. Run the GNU configure script
`# configure --prefix=/opt --enable-relocatable`
4. Build everything:
`# make`
or if you have a dual/multi core device and want to build faster:
`$ make -j #` where # is the number of cores you have
5. There is a little bug in the makefile that thinks there should be a shared version of one static-only library and then tries (and fails) to install it.  I poked around a little, but GNU configure scripts and especially their resultant makefiles are basically incomprehensible, so to work around the issue just fake the presence of the shared library by...
`# touch gettext-runtime/libasprintf/.libs/libasprintf.so.0.0.0`
6. Then you can install gettext...
`# make install`
7.  You can now just delete the fake shared libasprintf library that was installed
`# rm /opt/lib/libasprintf.so*

That should give you msgfmt and any other gettext tools you need.

zyxmon, 2021-12-15:
Entware (and Openwrt for sure) build а full set of binaries and libraries here - https://github.com/Entware/Entware/tree/master/package/libs/gettext-full
It is a matter of packaging msgfmt and extra libraries. I can upload all needed farm feeds) for testing.

zyxmon, 2021-12-15:
Which of the files - https://packages.debian.org/bullseye/amd64/gettext/filelist
are essential for packaging?