Entware/Entware #596: opkg segmentation fault FreshTomato w/ WNR3500L

mike406, 2021-01-30:
Hello,
I'm attempting to install from here: http://bin.entware.net/mipselsf-k3.4/installer/generic.sh

I get the following output, please advise what I'm doing wrong. I am able to successfully install the older version http://pkg.entware.net/binaries/mipsel/installer/installer.sh just fine however.

`Connecting to bin.entware.net (104.21.91.83:80)
-                    100% |****************************************************************************************************************************************************************************************************************************************************************************|  2760   0:00:00 ETA
Info: Checking for prerequisites and creating folders...
Warning: Folder /opt exists!
Info: Opkg package manager deployment...
Connecting to bin.entware.net (172.67.212.134:80)
opkg                 100% |****************************************************************************************************************************************************************************************************************************************************************************|   864k  0:00:00 ETA
Connecting to bin.entware.net (104.21.91.83:80)
opkg.conf            100% |****************************************************************************************************************************************************************************************************************************************************************************|   150   0:00:00 ETA
Info: Basic packages installation...
Segmentation fault
Segmentation fault
Info: Congratulations!
Info: If there are no errors above then Entware was successfully initialized.
Info: Add /opt/bin & /opt/sbin to $PATH variable
Info: Add "/opt/etc/init.d/rc.unslung start" to startup script for Entware services to start
Info: Found a Bug? Please report at https://github.com/Entware/Entware/issues
`

mike406, 2021-01-30:
Nevermind, found in the wiki that it is too old because MIPS 2.6.22 is not supported.

In that case then, is there any way I can use ngrok? That's really the only package I am looking to use.

zyxmon, 2021-01-31:
Try Entware-ng

closed 2021-01-31 by zyxmon

mike406, 2021-01-31:
> 
> 
> Try Entware-ng

No ngrok package in entware-ng. So I guess I am SOL