Entware/Entware #597: luci-app-sqm on entware

esunayg, 2021-01-31:
hello, i am a newbee to entware but have some basic knowledge about openwrt. recently I ve bought a "keenetic extra dsl" modem router. and I ve been able to install entware on it. but I just noticed there is no package related sqm on entware (correct me if I am wrong.) havent been able to find it on any package list. it was my only intention to install entware in the first place. and I am sure there are too many folks out there frustrated about buffer bloat. (gamer or not). currently I have been able to install lede on my old router linksys wrt1900ac and with sqm enabled on that, I have a very good dslreport score and I am very happy with it. but currently I have only one pc on my home so 2 routers looks very overkill & non energy efficient. my network is only 32mbits so I assume my modems cpu could easily handle it also. could you please make it or guide me to a start so I can try myself to compile for it (I know i cant but I am so motivated :) forgive me if there is double request but I have filtered and could find any) 
hope to hear from you and thanks for the good work!!!

zyxmon, 2021-01-31:
Entware is a repo of usermode binaries. QoS is supported by kernel. Please refer to cli description of your keenetic - something like http://docs.help.keenetic.com/cli/3.1/en/cli_manual_kn-2010.pdf or similar.

closed 2021-01-31 by zyxmon

esunayg, 2021-02-01:
thank you for help. I will read that. I also found the exact model pdf link if anyone needs.

http://docs.help.keenetic.com/cli/3.1/en/cli_manual_kn-2111.pdf