Entware/Entware #598: gpg: failed to start agent '/opt/bin/gpg-agent': No such file or directory

olivernovakovic, 2021-02-01:
**Describe the bug**
[gnupg2]: gpg2 --gen-key
leads to this:
...
gpg: failed to start agent '/opt/bin/gpg-agent': No such file or directory
gpg: can't connect to the agent: No such file or directory
gpg: agent_genkey failed: No agent running
Key generation failed: No agent running

**Environment (please complete the following information):**
 - Entware feed: [x64]
 - QNAP NAS
 - FW: 4.5.1.1540

zyxmon, 2021-02-02:
gpg2 in openwrt has 4 packages - https://github.com/Entware/entware-packages/blob/master/utils/gnupg2/Makefile#L139
3 packages are missing in the binary repo and will be added in near future.
You'll need to install gnupg2-utils and probably other packages.
PS Thanks for reporting. gpgv2 was first added to Entware and lately ported to openwrt and was split in 4 packages. We missed it.

closed 2021-02-02 by zyxmon

GwynethLlewelyn, 2023-08-16:
Dang. Same issue here, but on a Synology NAS. Note that, in my case, `gpg-agent` _does_ exist under `/bin` and _is_ running; however, `git` — which calls `gpg` to sign commits — gets utterly confused with the paths. The simplest fix seemed to be adding a symlink to the 'proper' `gpg-agent`.

Alas, it's not _that_ easy, since now I get the following error when testing to sign anything:
```bash
$ /opt/bin/gpg2 --status-fd=2 -bsau dd
gpg: can't connect to the agent: IPC connect call failed
gpg: keydb_search failed: No agent running XXXXXXXX
gpg: skipped "XXXXXXXX": No agent running
[GNUPG:] INV_SGNR 0 XXXXXXXX
[GNUPG:] FAILURE sign 3                                                                                                                                                         
gpg: signing failed: No agent running
```
(XXXXXXXX being the short form of the key)

This might be due to an incompatibility between whatever `gpg-agent` is installed system-wide, and the `gpg2` binary installed via opkg/Entware. I believe that this will only be 'fixed' once you release a version of `gpg-agent` that works fine with the supplied `pgp` executable (any news on that, after 2+ years?).

GwynethLlewelyn, 2023-08-16:
Oh, I forgot: in my case, `gpg-agent` complains about `--use-standard-socket`, and says it's an old option that has been ignored. I wonder — would that account for the `IPC connect call failed`, since that might be the call to a so-called "standard socket"? What is the recommended replacement these days, and how to enable it on both sides (e.g. `gpg` _and_ `gpg-agent`)?

GwynethLlewelyn, 2023-08-16:
Ok, please disregard my previous message(s). While all information I posted was essentially correct, I didn't notice a simple thing: because Entware apparently does not ship with its own `gpg-agent`, it was, indeed, using the system's own. Although both are at the same version (3.39 in my case), they were probably compiled/linked against different libraries, and that was sufficient to prevent them to talk to each other.

Reverting to the version that came with the system worked flawlessly, even inside Git.

zyxmon, 2023-08-16:
```
# opkg files gnupg2-utils
Package gnupg2-utils (2.2.39-1) is installed on root and has the following files:
/opt/bin/gpg-agent
/opt/bin/gpg-wks-server
/opt/sbin/applygnupgdefaults
/opt/bin/gpgparsemail
/opt/bin/gpgsm
/opt/lib/gpg-check-pattern
/opt/sbin/addgnupghome
/opt/bin/gpg-connect-agent
/opt/lib/gpg-wks-client
/opt/bin/gpgsplit
/opt/lib/gpg-protect-tool
/opt/bin/kbxutil
/opt/lib/gpg-preset-passphrase
/opt/bin/gpgtar

```

GwynethLlewelyn, 2023-08-18:
Thank you! I stand corrected — apparently, I didn't have the _whole_  `pgp2-utils` installed, and the mixing with the system-wide `gpg2` tools (even with all paths correctly set!) is prone to generate a terrible mess (thus the `/opt` directory!).