Entware/Entware #600: Support for mktorrent

adamwinn, 2021-02-03:
mktorrent is the gold standard when it comes to creating torrent files.  It's also faster than transmission-create

https://github.com/pobrn/mktorrent

- [x] No similar package already added and/or this one got some new killer feature(s),
- [x] Package is in active development (at least, not abandoned for 5 years or more),
- [x] I'm ready to test added package on my platform [e.g. mipselsf-k3.4] and get a feedback about it.

ryzhovau, 2021-02-04:
Please, use alternatives like `transmission-create` or `ctorrent-nossl`. I see no sense to add another package with similar purpose, sorry.

closed 2021-02-04 by ryzhovau

ryzhovau, 2021-02-14:
We've received another request of this tool and decided to add it to repo. Binaries will arrive soon.