Entware/Entware #601: Jamulus headless server

mjknowle, 2021-02-08:
**Describe the solution you'd like**

The "headless server" build of Jamulus 3.6.2 runs great on my Asus RT-AX86U for serving probably up to ~30 musicians. (I've tested with up to 12; each active client adds about 10-20% CPU use depending on settings). Jamulus is becoming increasingly popular during covid-19 as a platform for supporting small musical groups online, and the latency of running jamulus right on your router can be better than running a jamulus server inside your network (in my case, about 10-20msec, which is substantial).

See https://jamulus.io, and particularly, https://jamulus.io/wiki/Installation-for-Linux . I built with headless-server enabled, and am running (under a Debian environment on the router) with --fastupdate -s -n -T -u 30, and an added iptables rule to pop open the port to the public Internet.

- [x] No similar package already added and/or this one got some new killer feature(s),
- [x] Package is in active development (at least, not abandoned for 5 years or more),
- [x] I'm ready to test added package on my platform [e.g. mipselsf-k3.4] and get a feedback about it.

ryzhovau, 2021-02-08:
It's a cool idea on the first sight, but... I really don't understand why Jamulus should be installed on embedded devices with quite weak CPU. Latency is a top priority for everyone who ever tried to play music across internet (even duet with a teacher), so it's better to leave router with it network-related workload with no any additional tasks. And use something more powerful to run Jamulus.

mjknowle, 2021-02-12:
Jamulus is surprisingly CPU-lean (particularly as audio applications go) but very network-latency and jitter sensitive, and favours a consistent environment on whatever it's running on. A router nicely satisfies these properties, for small groups. To wit, people have been running the client (and presumably the server as well) on Raspberry Pi's successfully per https://sourceforge.net/p/llcon/discussion/hardware/thread/91475bcb82/. 

Reducing server latency -- giving a better experience for you and your fellow musicians -- is exactly *why* one would want to be able to run a small Jamulus server right on your router rather than having to maintain a separate machine for it--and pay the NAT/QoS/store-and-forward latency overhead. You're dealing with similar requirements to running a small asterisk conference bridge, basically, and that already has entware packages.

Having tested Jamulus successfully on my router for several sessions over the last couple of weeks, I can confirm that it's quite a viable option for small mostly-local-area groups, which judging from my experience and what's on the fora, is the usual use case.

ryzhovau, 2021-02-12:
Ok, I promise to look into.

ryzhovau, 2021-04-06:
[Requires](https://jamulus.io/wiki/Compiling) XCode build system, additional QT libraries and jack library, which is absent in our buildroot. Won't add, soory.

closed 2021-04-06 by ryzhovau